wlcrs/huawei_solar #666: [Bug]: Incorrect Total yield/day

wavelov3r, 2024-03-29:
### Describe the issue

The total yield energy  take in account only inverted energy, not the total (home battery recharge is excluded)

[image]
[image]

### Bescribe your Huawei Solar Setup

Inverter Type: 6ktl-l1
Inverter Firmware version: last
SDongle present:  no
Power meter present: single phase 
Battery: LUNA2000 15kWh / None
Battery Firmware version: last

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
No needs for logs
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2024-03-29:
Duplicate of #1

closed 2024-03-29 by wlcrs

wavelov3r, 2024-03-30:
I have set nothing. Also i've noticed time ago this bug, but i was thinking it was resolved with last release.

Today this is the result, repeat, no settings moved.

[image]
[image]