wlcrs/huawei_solar #670: Setup failed for custom integration 'huawei_solar': Unable to import component: cannot import name 'checkCRC' from 'pymodbus.utilities' (/usr/local/lib/python3.12/site-packages/pymodbus/utilities.py)

AnKloss, 2024-04-02:
### Describe the issue

Logger: homeassistant.setup
Source: setup.py:296
First occurred: 6:59:32 AM (1 occurrences)
Last logged: 6:59:32 AM

Setup failed for custom integration 'huawei_solar': Unable to import component: cannot import name 'checkCRC' from 'pymodbus.utilities' (/usr/local/lib/python3.12/site-packages/pymodbus/utilities.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 296, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 894, in async_get_component
    comp = self.get_component()
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 921, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/huawei_solar/__init__.py", line 17, in <module>
    from huawei_solar import HuaweiSolarBridge, HuaweiSolarException, InvalidCredentials, register_values as rv
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/__init__.py", line 6, in <module>
    from .bridge import HuaweiSolarBridge  # noqa
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/bridge.py", line 16, in <module>
    from .huawei_solar import DEFAULT_BAUDRATE, DEFAULT_SLAVE, DEFAULT_TCP_PORT, AsyncHuaweiSolar, Result
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 19, in <module>
    from pymodbus.utilities import checkCRC, computeCRC
ImportError: cannot import name 'checkCRC' from 'pymodbus.utilities' (/usr/local/lib/python3.12/site-packages/pymodbus/utilities.py)

### Bescribe your Huawei Solar Setup

Inverter Type:
Inverter Firmware version:
SDongle present: yes / no
Power meter present: three phase / single phase / no
Battery: LG RESU xxkWh / LUNA2000 xxkWh / None
Battery Firmware version:

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
Logger: homeassistant.setup
Source: setup.py:296
First occurred: 6:59:32 AM (1 occurrences)
Last logged: 6:59:32 AM

Setup failed for custom integration 'huawei_solar': Unable to import component: cannot import name 'checkCRC' from 'pymodbus.utilities' (/usr/local/lib/python3.12/site-packages/pymodbus/utilities.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 296, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 894, in async_get_component
    comp = self.get_component()
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 921, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/huawei_solar/__init__.py", line 17, in <module>
    from huawei_solar import HuaweiSolarBridge, HuaweiSolarException, InvalidCredentials, register_values as rv
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/__init__.py", line 6, in <module>
    from .bridge import HuaweiSolarBridge  # noqa
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/bridge.py", line 16, in <module>
    from .huawei_solar import DEFAULT_BAUDRATE, DEFAULT_SLAVE, DEFAULT_TCP_PORT, AsyncHuaweiSolar, Result
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 19, in <module>
    from pymodbus.utilities import checkCRC, computeCRC
ImportError: cannot import name 'checkCRC' from 'pymodbus.utilities' (/usr/local/lib/python3.12/site-packages/pymodbus/utilities.py)
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2024-04-02:
Another pymodbus version is installed than what the integration supports. This happens when two or more integrations use that same dependency, but require different versions. v1.3.4b1 requires pyModbus v3.6.6

You will need to make sure that all modbus-dependent integrations support the same pymodbus version. Welcome in this special place of hell called the "Python dependency hell".

ThomDietrich, 2024-04-03:
Correction: Reinstalling version 1.4.0a1 solved the issue for me.

closed 2024-04-16 by wlcrs