wlcrs/huawei_solar #671: [Bug]:

Ivano62, 2024-04-03:
### Describe the issue

With latest version on HA 2024.04 I get this error

2024-04-03 23:43:42.067 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up huawei_solar platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 356, in _async_setup_platform
    await asyncio.shield(awaitable)
  File "/config/custom_components/huawei_solar/sensor.py", line 808, in async_setup_entry
    assert ucs.power_meter_update_coordinator
AssertionError

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-6KTL-L1
Inverter Firmware version: V200R001C00SPC136
SDongle present: yes
Power meter present: single phase
Battery: LUNA2000 15kWh
Battery Firmware version: V100R002C00SPC127

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
2024-04-03 23:43:42.067 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up huawei_solar platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 356, in _async_setup_platform
    await asyncio.shield(awaitable)
  File "/config/custom_components/huawei_solar/sensor.py", line 808, in async_setup_entry
    assert ucs.power_meter_update_coordinator
AssertionError
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

mariusnauj, 2024-04-04:
Hello,

Which version of integration you are using?

closed 2024-04-04 by wlcrs

wlcrs, 2024-04-04:
Thank you for testing the alpha release. Can you please test again with 1.4.0 alpha 2 ?

commit 347054f6 (2024-04-04): Fix power meter detection logic 

Fixes #671