wlcrs/huawei_solar #683: Set max power to grid (register 40125)

haasje30, 2024-04-15:
### Describe your feature request

Hello, 

Max grid power which is implemented at this moment is concerning battery connected inverter. I have no batteries, but would like to limit power from the inverter to the grid. This can be done with register 40125, and I have been experimenting with the modbus integration in HomeAssistant, but was not able to write registers. (reading was no problem). So it would be nice if the integration could write register 40125, or if someone could explain how to write registers with the modbus integration in HA. (probaly a sort of login is required?)

Thanks.

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

Roving-Ronin, 2024-04-16:
@haasje30 

Have you looked at the existing services....

[image]

closed 2024-04-16 by wlcrs

wlcrs, 2024-04-16:
I've just released 1.4.0 alpha 3 containing 2 new entities corresponding with register 40125 and 40126. 

As both seem to influence the active power output, I've disabled the percentage-based derating by default. Only the absolute value register 40126 is visible by default as I find it easier to understand what it means.