wlcrs/huawei_solar #696: [Bug]: Failed Setup using 2024.4.b0

yousaf465, 2024-04-29:
### Describe the issue

Since updating to HA 2024.4.b0 Huwaei solar isn't working. with error

This error originated from a custom integration.

Logger: pymodbus.logging
Source: custom_components/huawei_solar/__init__.py:80
integration: Huawei Solar (documentation, issues)
First occurred: 11:35:06 (12 occurrences)
Last logged: 11:41:37

Failed to connect

### Describe your Huawei Solar Setup

Inverter Type: SUN2000-10KTL-M1
Inverter Firmware version:
SDongle present: yes
Power meter present: three-phase
Battery:  None
Battery Firmware version: 
Huawei Solar integration version: 1.3.3

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

[home-assistant_huawei_solar_2024-04-29T06-40-43.964Z.log](https://github.com/wlcrs/huawei_solar/files/15145905/home-assistant_huawei_solar_2024-04-29T06-40-43.964Z.log)

### Upload your relevant debug logs

```text
Logger: huawei_solar.huawei_solar
Source: /usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py:185
First occurred: 11:46:30 (5 occurrences)
Last logged: 11:49:40

Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/locks.py", line 212, in wait
    await fut
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 179, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 108, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 242, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 278, in get_multiple
    response = await self._read_registers(registers[0].register, total_length, slave)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 398, in _read_registers
    async with self._communication_lock():
  File "/usr/local/lib/python3.12/contextlib.py", line 210, in __aenter__
    return await anext(self.gen)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 146, in _communication_lock
    raise err
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 142, in _communication_lock
    await asyncio.wait_for(self._client.connected_event.wait(), WAIT_FOR_CONNECTION_TIMEOUT)
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 519, in wait_for
    async with timeouts.timeout(timeout):
  File "/usr/local/lib/python3.12/asyncio/timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

Logger: huawei_solar.huawei_solar
Source: /usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py:185
First occurred: 11:46:30 (5 occurrences)
Last logged: 11:49:40

Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/locks.py", line 212, in wait
    await fut
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 179, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 108, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 242, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 278, in get_multiple
    response = await self._read_registers(registers[0].register, total_length, slave)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 398, in _read_registers
    async with self._communication_lock():
  File "/usr/local/lib/python3.12/contextlib.py", line 210, in __aenter__
    return await anext(self.gen)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 146, in _communication_lock
    raise err
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 142, in _communication_lock
    await asyncio.wait_for(self._client.connected_event.wait(), WAIT_FOR_CONNECTION_TIMEOUT)
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 519, in wait_for
    async with timeouts.timeout(timeout):
  File "/usr/local/lib/python3.12/asyncio/timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

This error originated from a custom integration.

Logger: pymodbus.logging
Source: custom_components/huawei_solar/__init__.py:90
integration: Huawei Solar (documentation, issues)
First occurred: 11:46:25 (5 occurrences)
Last logged: 11:49:35

Failed to connect
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

tmenguy, 2024-04-29:
Hi, strange I'm on the 2024.4.4 official and it works great

MarjoCo, 2024-04-29:
I just updated the 2024.4.4 and it stopped working for me. I had version 1.3.1 working prior updating to 2024.4.4. Then updated to 1.3.3 and it resolved the issue.

wlcrs, 2024-04-29:
This issue seems to be caused by a connection failure or an unstable connection to the inverter. This integration already has already implemented a lot of logic to automatically recover from connection failures or unstable connections. The three most important ones are:
- automatic reconnects if the connection to the inverter failed;
- each Modbus request is retried for up to 6 times, with increasing waits between each attempt to allow the connection to recover;
- after 3 failed Modbus requests the connection is reset to fix hanging connections.

Please understand that I cannot help you solve your connection failure or unstable connection. Consider [an alternative connection method](https://github.com/wlcrs/huawei_solar/wiki/Connecting-to-the-inverter).
If you are convinced that there is no connectivity issue, then I invite you to debug and fix the problem yourself and submit a pull request to the underlying [huawei-solar Python library](https://gitlab.com/Emilv2/huawei-solar). Your input would be greatly appreciated.

As there is nothing actionable for me to do in this issue, I'm closing it. You are welcome to open a topic in [Discussions: Connecting to the inverter](https://github.com/wlcrs/huawei_solar/discussions/categories/connecting-to-the-inverter) to get community support for your problem.

tmenguy, 2024-04-29:
And failed to say : I’m on latest beta

yousaf465, 2024-04-30:
The issue was solved by restarting the wifi module of my wifi router.