wlcrs/huawei_solar #698: [Feature Request]: Inverter's' Total Sensors

Roving-Ronin, 2024-05-01:
### Describe your feature request

Hi Thijs,

Just wondering if it might be possible to add an 'Inverters' function, same as you've added the 'Batteries' function in v1.4.0x? (& Disabled by default).

If this is possible, this would negate the need for users with multiple inverters to have to manually create custom sensors that sum up the Inverters:
- Active Power
- Input Power
- Total Yield
- Daily Yield

This would be beneficial by ensuring standardisation of these sensors and their entity ID, whilst also allowing for their use by 3rd party addon/card etc (i.e. copy/paste configurations for cards can be published, as everyone is now using the same default sensors id) 

My understanding is that, given the modbus master heirarchy, this summed up information is to be sourced from the sDongle. Looking at the sDongle Modbus Interface Definitions ( https://photomate.zendesk.com/hc/en-gb/article_attachments/13800172732829 ), it appears that at least some of these combined values, such as 'Total Input Power' & 'Total Active Power', are available from:

[image]

Thanks for your consideration.

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

Roving-Ronin, 2024-05-09:
Thijs, any chance to look at this?

wlcrs, 2024-05-09:
Not trivial to implement. I don't own such a dongle, so I cannot test, making it very hard to get everything right. And that's the case for both for the initial implementation and subsequent maintenance of this feature.

So no, I'm not going to invest time into this one. Pull requests are welcome. If anyone is interested in implementing this, feel free to open a PR in a very early stage so that we can discuss the best approach.

Final note: the 'Batteries' entities will only cover the batteries connected to 1 inverter. If you have LUNA batteries spread over multiple inverters, this will create multiple 'Batteries' entities in HA.

Roving-Ronin, 2024-05-10:
Hi Thijs,

Just wondering, what is your personal Huawei setup that you have to test and develop against? I had assumed some L1 / M1 and LUNA ESS (that each come with a bundled sDongle incldued), but sounds like you might have an L0 with onboard WiFi AP ?

Re the Batteries, thanks for that clarification. As I understand it currently, the integration will support 2 batteries (BMS) per inverter. However there's a 'future' issue that the new LUNA S1 model scales to 4 batteries per inverter (i.e. 4 x 21kWh = 84kWh of batteries per inverter.... possible after winning the Lotto).

Currently though, if someone where to have 2 x LUNA ESS and have them attached one to each of two inverters (or if they had 3-4 LUNA ESS, so multiple attached to 2 inverters), would you know who the HA entities would be named? i.e. Inverter 1 would have batteries_charge_discharge_power, so would it be Inverter 2 would have a batteries_charge_discharge_power_2 (and likewise for the other batteries sensors) ?

IMHO strange that I know there's people in your part of the world that are made beta users/sites and provided with access to new products to test prior to market release, you'd think that Huawei Sales / Marketing would seek to engage with yourself and make you a beta user, as it will only help to drive their products adoption / help drive sales.

PS. Is it worth converting this Issue to a Discussion for 'Inverters' ?

Roving-Ronin, 2024-06-10:
.

dertuerke, 2024-06-16:
Push

Roving-Ronin, 2024-07-17:
> Pull requests are welcome. If anyone is interested in implementing this, feel free to open a PR in a very early stage so that we can discuss the best approach.

Hi Thijs,
Could you give any guidance as to what would be required? trying to 'follow the logic' of the coding to work out how to do it.

Roving-Ronin, 2024-08-17:
.

Roving-Ronin, 2024-09-18:
.

closed 2024-10-29 by github-actions[bot]