wlcrs/huawei_solar #700: [Bug]: polling frequence in the latest beta release

jodewee, 2024-05-03:
### Describe the issue

Not really a bug, but a question;
do you have an example of this one?
[image]

I now have this automation for the polling;
[image]

### Describe your Huawei Solar Setup

Inverter Type: sun2000 4.6ktl L
Inverter Firmware version:
SDongle present: yes 
Power meter present: single phase 
Battery: LUNA2000 10kwh
Battery Firmware version:
Huawei Solar integration version: latest beta

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
no issues on the integration
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2024-05-03:
No, I haven't tested this at all to be honest. I just expect this to be needed due to the code changes.

Please test and let me know if your current approach works and behaves like expected or not.

jodewee, 2024-05-03:
well, to me everything seems to be working with the automation above

alexgmp9, 2024-05-15:
Can confirm that it works well. At 10 seconds, for me. Under 10 seconds it seems that some requests are missed.

catalinbordan, 2024-06-11:
Hi, Also I have the same automation on wifi connection, but the response it is at 10 seconds, even if I put 5 seconds. I was thinking that I need an Sdongle if I want shorter update intervals...
@jodewee can you please confirm that your sensors are updated according to your automation ~5 seconds?

alexgmp9, 2024-06-11:
I have two sun2000-4ktl-l1 cascaded. The internet connection is via wlan using an SDongle. 
I used debugging and the pooling of data takes 2-5 seconds for each inverter. I tried to change the pooling time to 5 seconds but i get a lot o timeouts and the inverters get unavailable. I have not tried yet lan connection. Maybe the it is faster. 
For single invertor setup the 5 seconds pooling seems reasonable.

jodewee, 2024-06-11:
@catalinbordan , the automation polls every 5sec, but the data itself seems to be only changing once every 10sec.
Not that of a deal for me, it is a lot better than the 2min average of the huawei portal 😀

catalinbordan, 2024-06-11:
@alexgmp9 thanks for the info. Just read again the wiki documentation. thumb up for the detailed how to do. I see that they do not recommend Sdongle. In fact I am a little bit confused about sDongle types. SUN2000L I think has Sdongle WiFi-only by factory witch I believe I'm using by now by connecting HA through internal network of router(by router dhcp ip released). We also have possibility to add another sDongle with port LAN cable(recommended for >2 inverters) So at the end we will have 2 sDongle, witch the last one (LAN) must be configured to be something like master connection for communications with fusion solar server and also activate local O&M/modbus.
So @alexgmp9 what kind of sDongle you have? :) because you are saying wlan(wireless lan), but also you are talking about 2 inverters (I know it is possible to connect 2 inverter in master/slave with Wi-Fi dongle). People prefer to use LAN cable because Wi-Fi connection is unstable, but I don't have issues(I also changed with a bigger antenna the inverter's antenna). I want to buy LAN sDongle only if I will get more quickly and reliable data updates(to difficult to get the LAN cable to inverter). Do you have the possibility to test it on LAN?

@jodewee thanks for the update. So just like me :). Could you please tell me if you are connection to inverter by LAN cable?

alexgmp9, 2024-06-11:
@catalinbordan I have an SDongle A-05 which is connected to the master inverter, and the slave is cascaded using RS485 connection. Even though the inverters have built-in wifi, it is not used to connect the inverter to internet. Only the SDongle in connected to internet via wifi. The SDongle manages the two inverters and the communication to the cloud or to HA. This is why I can pool data from both inverters at the same time. 
The WLAN FE SDongle can be connected either through wifi or trough wired connection (see attached image). 
You can buy one and add it to your sistem.  In RO it is about 250lei (50euro).
[image]

I have not tried yet the lan connection due to lack of a long enough wire (I will buy one in the near future) and due to lack of time. I will try this type of connection and come back with feedback as soon as possible.

catalinbordan, 2024-06-11:
Here it is a possible cause and a solution:
https://github.com/wlcrs/huawei_solar/issues/532

jodewee, 2024-06-12:
@catalinbordan i have an AP that act as a bridge between my wired network and the sun2000's AP