wlcrs/huawei_solar #704: [Bug]: 1.4.0a3 - sensor.power_meter_reactive_energy - state class total_increasing, but its state is negative

Roving-Ronin, 2024-05-09:
### Describe the issue

Hi Thijs,

Just noticed this message in the log files of a remote site running 1.4.0a3, appears to be a potential issue with the choice of state_class, currently 'total_increasing'... looks like 'meaurement' might be more applicable?

Logger: homeassistant.components.sensor.recorder
Source: components/sensor/recorder.py:325
integration: Sensor (documentation, issues)
First occurred: 12:52:00 am (1 occurrences)
Last logged: 12:52:00 am

Entity sensor.power_meter_reactive_energy from integration huawei_solar has state class total_increasing, but its state is negative. Triggered by state -7155.54 with last_updated set to 2024-05-08T14:39:59.999999+00:00. Please create a bug report at https://github.com/wlcrs/huawei_solar/issues

### Describe your Huawei Solar Setup

Inverter Type:    2 x L1 5kW
Inverter Firmware version:  V200R001C00SPC138
SDongle present: Yes
sDongle Firmware: V200R022C10SPC108
Power meter present: single phase 
Battery: 1 x LUNA2000 10kWh 
Battery Firmware version: V100R002C00SPC125
Huawei Solar integration version: 1.4.0a3

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[config_entry-huawei_solar-a1ed6e11e2ae4394800af6c28a10e35e.json](https://github.com/wlcrs/huawei_solar/files/15256488/config_entry-huawei_solar-a1ed6e11e2ae4394800af6c28a10e35e.json)

### Upload your relevant debug logs

```text
N/A. Potential State_Class definition error only.
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2024-05-09:
Is this a steadily decreasing negative number? In that case 'total' instead of 'total_increasing' may be a better choice than measurement.

Roving-Ronin, 2024-05-09:
Appears to be a negative number, that is slowly decreasing.

Last 6 hours:
[image]

Last Week: 
[image]

commit 1c436353 (2024-05-09): Make GRID_ACCUMULATED_REACTIVE_POWER a measurement

Fixes #704
closed 2024-05-09 by wlcrs