wlcrs/huawei_solar #706: [Bug]: Integration fails to load

carlhye, 2024-05-10:
### Describe the issue

The integration fails to load.
Redownloading the HACS integration does not solve this
Updating everything has not solved it
Restarting HA foes not solve this

### Describe your Huawei Solar Setup

Inverter Type: SUN2000-3KTL-M1
Inverter Firmware version: V100R001C00SPC153
sDongle present: Yes
sDongle Type: sDongleA-05 (WiFi / Ethernet)
sDongle Connectivitiy: Ethernet
sDongle Firmware: V100R001C00SPC125
Power meter present: three phase
Optimizers Present: No
Battery: LUNA2000-SO 15kWh / LUNA2000-S1 15kWh (Released 2024) - unknown if SO or S1
Battery Firmware version: N/A - can't see it in the app
Huawei Solar integration version: 1.3.3

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

Option is not there

### Upload your relevant debug logs

```text
Logger: homeassistant.setup
Source: setup.py:322
First occurred: 10:00:13 (1 occurrences)
Last logged: 10:00:13

Setup failed for custom integration 'huawei_solar': Unable to import component: cannot import name 'checkCRC' from 'pymodbus.utilities' (/usr/local/lib/python3.12/site-packages/pymodbus/utilities.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 981, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1041, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 144, in protected_loop_func
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/huawei_solar/__init__.py", line 17, in <module>
    from huawei_solar import HuaweiSolarBridge, HuaweiSolarException, InvalidCredentials, register_values as rv
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/__init__.py", line 6, in <module>
    from .bridge import HuaweiSolarBridge  # noqa
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/bridge.py", line 16, in <module>
    from .huawei_solar import DEFAULT_BAUDRATE, DEFAULT_SLAVE, DEFAULT_TCP_PORT, AsyncHuaweiSolar, Result
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 19, in <module>
    from pymodbus.utilities import checkCRC, computeCRC
ImportError: cannot import name 'checkCRC' from 'pymodbus.utilities' (/usr/local/lib/python3.12/site-packages/pymodbus/utilities.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 322, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1001, in async_get_component
    self._component_future.result()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 993, in async_get_component
    comp = self._get_component()
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1041, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 144, in protected_loop_func
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/huawei_solar/__init__.py", line 17, in <module>
    from huawei_solar import HuaweiSolarBridge, HuaweiSolarException, InvalidCredentials, register_values as rv
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/__init__.py", line 6, in <module>
    from .bridge import HuaweiSolarBridge  # noqa
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/bridge.py", line 16, in <module>
    from .huawei_solar import DEFAULT_BAUDRATE, DEFAULT_SLAVE, DEFAULT_TCP_PORT, AsyncHuaweiSolar, Result
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 19, in <module>
    from pymodbus.utilities import checkCRC, computeCRC
ImportError: cannot import name 'checkCRC' from 'pymodbus.utilities' (/usr/local/lib/python3.12/site-packages/pymodbus/utilities.py)

Logger: homeassistant.util.loop
Source: util/loop.py:70
First occurred: 10:00:13 (1 occurrences)
Last logged: 10:00:13

Detected blocking call to import_module with args ('custom_components.huawei_solar',) in /usr/src/homeassistant/homeassistant/loader.py, line 1041: ComponentProtocol, importlib.import_module(self.pkg_path) inside the event loop
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

dngroza, 2024-05-10:
I have the same problem with version 1.3.3. I have another HA instance with 1.3.3 version and it's working fine.

carlhye, 2024-05-13:
Just talked to the installer of the plant, and he's initiating a firmware update of the dongle to V2, so let's see if this solves it.

What I find strange is, that it seems like the integration is not even loading - if I remove the integration and reinstall it, would all device entities be the reloaded or would now ones be created?

Misiu, 2024-05-20:
@carlhye it's a problem with Python (the programming language Home Assistant is written in), ref: https://github.com/wlcrs/huawei_solar/issues/661
Temporary solution: update integration to 1.3.4b1 (this is a beta release)

wlcrs, 2024-05-23:
Duplicate of #661

closed 2024-05-23 by wlcrs