wlcrs/huawei_solar #709: [Bug]: Unable to setup Huawei Solar integration

MrErichsen, 2024-05-12:
### Describe the issue

After succesfull download, I receive the error: "message":"Invalid handler specified"}

Looking into the log, the following is noted:
Error occurred loading flow for integration huawei_solar: cannot import name 'checkCRC' from 'pymodbus.utilities' (/usr/local/lib/python3.12/site-packages/pymodbus/utilities.py)

I am completely new to HA and I believe I have found the relevant log entries. If I need to supply more info, please be very detailed in your guidance :-)

### Describe your Huawei Solar Setup

Inverter Type: 
Inverter Firmware version: 
sDongle present: Yes / No
sDongle Type: sDongleA-05 (WiFi / Ethernet) / SDongleA-03 (4G)
sDongle Connectivitiy: WiFi / Ethernet / 4G
sDongle Firmware: 
Power meter present: three phase / single phase / no
Optimizers Present: Yes / No
Battery: LUNA2000-SO xxkWh / LUNA2000-S1 xxkWh (Released 2024) / LG RESU xxkWh / None
Battery Firmware version: 
Huawei Solar integration version: 

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
Error occurred loading flow for integration huawei_solar: cannot import name 'checkCRC' from 'pymodbus.utilities' (/usr/local/lib/python3.12/site-packages/pymodbus/utilities.py)
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

Lutz88, 2024-05-13:
Same for me!

Happend after updating the inverter, MBUS and dongle to the actual firmware tonight.

Integration can not start again and a new setup is not possible. Same error message like above.

Thanks for your help!

carlhye, 2024-05-13:
I get pretty much the same issue - also raised an issue a couple day ago here: https://github.com/wlcrs/huawei_solar/issues/706

MrErichsen, 2024-05-15:
Updated to 1.4.0a3 and the issue was fixed

closed 2024-05-15 by MrErichsen