wlcrs/huawei_solar #713: [Bug]: v1.4.0a3 - MAJOR - Delete Plant and Huawei Service Removed (and won't reinstall)

Roving-Ronin, 2024-05-21:
### Describe the issue

Needed to delete the plant within HA / WLCRS integration. Did so and rebooted HA, then proceeded to add my plant/inverters back into HA using the WLCRS integration, then rebooted.

Checked in Developer Tools / Services and discovered that ALL the Huawei services are missing.

Repeated and same outcome. Then uninstalled WLCRS integration itself and rebooted HA and tried reinstalling, with the same outcome. Repeated above and reinstalled v1.3.3 of the intetgration only to again find that the Huawei services are no longer available.

Eventually had to recover using a Backup from a week ago. Following hours doing this and updating config changes made since then, found that WLCRS is working (v.1.4.0a3 again and services listed). Deleted plant/inverters from HA Devices and confirmed that this (once again) sees the Huawei services removed. As per above, attempting to add the plant/inverters doesn't re-add the services, and again removing and reinstalling WLCRS fails to also reinstall them.

### Describe your Huawei Solar Setup

Inverter Type: 2 x SUN2000-6KTL-M1
Inverter Firmware version:  V100R001C00SPC160
sDongle present: Yes
sDongle Type: sDongleA-05 (WiFi / Ethernet)
sDongle Connectivitiy: WiFi 
sDongle Firmware: V200R022C10SPC110
Power meter present: 3 phase
Optimizers Present: Yes x 38
Battery: None
Battery Firmware version: N/A.
Huawei Solar integration version:  v1.4.0a3

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

[config_entry-huawei_solar-783cdf7acf39d956a5fb18b1abd63f76.json](https://github.com/wlcrs/huawei_solar/files/15387680/config_entry-huawei_solar-783cdf7acf39d956a5fb18b1abd63f76.json)

### Upload your relevant debug logs

```text
Core               2024.5.2  (tried 5.4 also, before restoring back to 5.2)
Supervisor         2024.05.1
Operating System   12.3
Frontend           20240501.1

[home-assistant_huawei_solar_2024-05-21T09-22-13.539Z.log](https://github.com/wlcrs/huawei_solar/files/15387642/home-assistant_huawei_solar_2024-05-21T09-22-13.539Z.log)
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2024-05-23:
According to your diagnostics file, the "advanced: elevated permissions" checkbox was not checked during setup?

Roving-Ronin, 2024-05-24:
🤦‍♂️ yes I had that off as also trying to stop the optimisers being included, as with 38 it takes around 2.5min for the integration to start every time it's reloaded / rebooted.

Is there any other way to enable advanced permissions, but not include optimisers.

Apologies on my brain fart and not putting 2+2 together for the services.