wlcrs/huawei_solar #714: [Bug]:  Grid-Connected, Grid connection with derating due to power rationing

JohnMcLear, 2024-05-21:
### Describe the issue

When using FusionSolar & Dongle Inverter works fine.  When using this integration I am unable to effectively use my PV/battery storage due to:

``Grid-Connected, Grid connection with derating due to power rationing ``

STR:
1. Remove Dongle.
2. Install EW11
3. Install This integration
4. Connect via Network > ew11.lan port 8899 slaves 6,2,3
5. See all inverters/batteries/smart meter in HA.

Screen shots:
![all devices working]https://github.com/wlcrs/huawei_solar/assets/220864/6d19fc9b-3f3c-495f-b016-31b90b06c356)
[image]
[image]

I'm not sure if the battery abnormal is related, the battery functions fine and has no errors/faults when using FusionSolar..

### Describe your Huawei Solar Setup

Inverter Type: 3x SUN2000-6KTL-L1 - See Daisy chaining documentation
Inverter Firmware version: V200R001C00SPC139
sDongle present: No, I'm using an EW11
Power meter present: single phase
Optimizers Present: No
Battery: 9x LUNA2000-SO 5kWh
Battery Firmware version: V100R002C00SPC127
Huawei Solar integration version: ???

### How do you connect to the inverter?

Over serial, with a serial-to-WiFi stick

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[config_entry-huawei_solar-cf67b76e647d9c7bbd8e02d643932326.json](https://github.com/wlcrs/huawei_solar/files/15387990/config_entry-huawei_solar-cf67b76e647d9c7bbd8e02d643932326.json)

### Upload your relevant debug logs

```text
No errors in logs when doing debug and running command...

cat home-assistant_huawei_solar_2024-05-21T09-47-56.135Z.log | grep huawei
```
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

JohnMcLear, 2024-05-21:
Update: I called the service ``huawei_solar.reset_maximum_feed_grid_power`` and it appears to have put the inverters back to normal on-grid mode.

I don't know why I needed to call that service: Ref: https://github.com/wlcrs/huawei_solar/wiki/Changing-Active-Power-Control

Roving-Ronin, 2024-05-21:
@JohnMcLear I've seen this before on a users plant (2 x 5kW L1's) where the installer had set the 'Grid Code' on the inverters to the newest version of said codes for Australian users (i.e. Australia-AS4777_<A/B/C>_LV230). These new codes are less tolerant than the settings the prior codes (AS4777-<NSW/QLD etc>) applied, so normal operation for the inverters always runs with this message.  Battery operation was NOT affected by this message.

[image]

I note that your 'Device Status' for the inverter though has a message that IMHO I'd been looking more into (than the Inverter State) message.   i.e. "Grid Connection: Power Limited".

JohnMcLear, 2024-05-21:
I have noticed that the batteries are NOT charging as they  should be..

Despite having enough power input.

[image]

The batteries are not charging as expected.

[image]

And instead the power is being fed to the grid.

[image]

JohnMcLear, 2024-05-21:
I'm also seeing a lot of items in the Logbook that I wasn't experiencing in Fusion Solar..

[image]

JohnMcLear, 2024-05-21:
> @JohnMcLear I've seen this before on a users plant (2 x 5kW L1's) where the installer had set the 'Grid Code' on the inverters to the newest version of said codes for Australian users (i.e. [...]

I'm using the same Grid Code that was set when I was using Fusion Solar.

JohnMcLear, 2024-05-22:
At midnight the inverters decided to stop discharging the batteries to help w/ load on the grid.

At 8am the inverters decided to send most of my energy storage to the grid.

[image]

[image]
[image]
[image]

Note the grid overvoltage error..

JohnMcLear, 2024-05-23:
After updates still no progress.  For some reason I get a battery abnormal when using this plugin but not fusion solar.  I think my next step should be to do a full reset of all three inverters and setup from scratch..

I'm going to wait for wlcrs before I do that tho because I feel like other people could hit this issue when migrating.

wlcrs, 2024-05-23:
Other reports of users with "locking" inverters were mostly due to the grid voltage getting too high (due to too many solar panel installations in the neighbourhood) and the inverter scaling back or even stopping production due to that. So maybe check your grid voltage readings to see if that is also the cause for your case?

Possibly the SDongle does something to balance your 3 inverters to mitigate the risk of having to scale back due to "overproduction" ?

I've always tried to make sure that using this integration has as little as unintended side-effects as possible. I do not change any settings without user interaction. So if you don't change any values in HA yourself, the integration will not change anything on the inverters/batteries.

JohnMcLear, 2024-05-26:
For now I've switched back to FusionSolar.

I think this requires further time/effort and maybe @wlcrs needs equipment IE multiple inverters/batteries to replicate the issue?

Roving-Ronin, 2024-05-26:
@JohnMcLear 
As Thijs mentioned, this is 100% to do with the grid settings that are applied by the settings within the inverter (and/or via FusionSolar).
I'd be checking g your grid settings first to confirm they have been correctly applied for the country / network your on.

malakudi, 2024-06-07:
"Grid-Connected, Grid connection with derating due to power rationing" is not a problem,, it is an indication that inverter does not work at 100% because it either has limitations in grid export (zero export or other limits) or limits due to high voltage of the grid etc
2068: Battery Abnormal is a bug with firmware V200R001C00SPC138 and later when also using Dongle. I had the same issue I flashed again  V200R001C00SPC136 and problem resolved. I haven't yet tried V200R001C00SPC141 if problem is fixed but I doubt it, I have reported issue to Huawei but they haven't responded  yet, even though I told them issue is fixed when rolling back to older firmware.

JohnMcLear, 2024-06-10:
Hey @malakudi thanks for your comment.

@malakudi I don't experience derating when using the dongle and FusionSolar

@malakudi 2068 was being experienced **without** the Dongle and I haven't seen it since I migrated back to FusionSolar.

@Roving-Ronin yeah I will re-try the integration and come back to you :)  Thanks for that suggestion.

malakudi, 2024-06-12:
> Thanks all for your input
> 
> @malakudi I don't experience derating when using the dongle and FusionSolar
> 
> @malakudi 2068 was being experienced **without** the Dongle and I haven't seen it since I migrated back to FusionSolar.
> 
You do experience it, fusionsolar might not show it but if you check what is reported through modbus, it is what this integration shows.
Error 2068 does not show without dongle. Error is generated on inverter reboot. Have you rebooted your inverters recentrly? With 139 firmware + dongle, you will get the error with or without the integration.

wlcrs, 2024-06-15:
I've added a link to this issue in the first section of https://github.com/wlcrs/huawei_solar/wiki/Connecting-to-the-inverter . I'm closing this issue as it is not a bug in the integration, but a limitation of the Huawei SUN2000 / LUNA2000 systems.

closed 2024-06-15 by wlcrs

spacemanspiff2007, 2024-11-03:
@JohnMcLear 
Can you try setting the inverter to a power limit (e.g. 70%) to the fusion solar app or portal and then back to unlimited?
It seems to show unlimited and only when setting to a limit and then back to unlimited the value is actually used.