wlcrs/huawei_solar #715: [Feature Request]: Entities with selectable entries

spelzl, 2024-05-21:
### Describe your feature request

Please have a look at how the entities of the go-e charger HA integration look like with editable number fields and pull down sections. Could the Huawei HA integration have similar entities, e.g. for battery max discharger power?
[image]

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

Roving-Ronin, 2024-05-21:
@spelzl 

You mean like this one that is documented...

[image]

spelzl, 2024-05-21:
Hi @Roving-Ronin,
I run 1.3.3.
I can't find what you described:
[image]

spelzl, 2024-05-21:
@Roving-Ronin do you have an idea when 1.4.0 will be officially released?
It appears as if 1.4.0a3 is close to become released soon.

Roving-Ronin, 2024-05-21:
@spelzl only Thijs (WLCRS) knows release timeframes. V1.4.0a3 has been out for a while now, next is supposed to be v1.4.0a4

wlcrs, 2024-05-23:
As stated in https://github.com/wlcrs/huawei_solar?tab=readme-ov-file#network-configuration:

Checking the Advanced: elevate permissions checkbox will:

- give you access to optimizer data
- **enable you to dynamically change your inverter and battery settings**

closed 2024-05-23 by wlcrs