wlcrs/huawei_solar #720: [Bug]: Error between Battery and Batteries Charge/Discharge Power sensor values

Roving-Ronin, 2024-05-28:
### Describe the issue

System has a LUNA 10kW ESS attached to the primary inverter, using the new 'Batteries Charge/Discharge Power' sensor for readings and noticing that the reading from it are different from the 'Battery 1 Charge/Discharge Power' sensor.

[image]

These readings above are static for a while, so its not one sensor being behind the other in getting an update. Constantly seeing one giving a positive value whilst the other is showing negative.

PS. Sorry not my battery, one 5000km away I have access to, otherwise I'd turn on the kettle etc to make the above numbers bigger to highlight the gap.

### Describe your Huawei Solar Setup

Inverter Type: 2 x L1 5kW inverters cascaded. Currently only primary inverter is operating (as it has a LUNA ESS attached)
Inverter Firmware version:  V200R001C00SPC138
sDongle present: Yes 
sDongle Type: sDongleA-05 
sDongle Connectivitiy: WiFi 
sDongle Firmware: V200R022C10SPC108
Power meter present: Single Phase 
Optimizers Present: No
Battery: LUNA2000-SO 10kWh
Battery Firmware version: V100R002C00SPC125
Huawei Solar integration version: v1.4.0a3

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[config_entry-huawei_solar-e7190a773e4143ed923016cd2d3a7e21.json](https://github.com/wlcrs/huawei_solar/files/15470214/config_entry-huawei_solar-e7190a773e4143ed923016cd2d3a7e21.json)

### Upload your relevant debug logs

```text
Page error, can't get debug to upload. Will re-try after submitting.
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2024-06-03:
Nothing guarantees that those values are synchronized when being read from the inverter. So this is not a bug.

A clear case of Segal's Law:

> A man with a watch knows what time it is. A man with two watches is never sure.

This was also the reason that I didn't export those values in earlier versions of my integration 🙃

closed 2024-06-03 by wlcrs

Roving-Ronin, 2024-06-03:
NP, just flagging for visibility. Having it raised with Huawei developers also.