wlcrs/huawei_solar #736: [Bug]:

Thomsen-79, 2024-06-16:
### Describe the issue

unable to Limit the power fed to the grid in homeassistant

### Describe your Huawei Solar Setup

Inverter Type: 5ktl-m1 and 15ktl-m2
Inverter Firmware version: V100R001C00SPC164
sDongle present: Yes 
sDongle Type: sDongleA-05 (WiFi / Ethernet) 
sDongle Connectivitiy: WiFi 
sDongle Firmware: V200R022C10SPC114
Power meter present: three phase - YES
Optimizers Present: No
Battery:  None
Battery Firmware version: 
Huawei Solar integration version: 

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[config_entry-huawei_solar-78aec80e4426a533758e89cc36aa70fa.json](https://github.com/user-attachments/files/15857267/config_entry-huawei_solar-78aec80e4426a533758e89cc36aa70fa.json)

### Upload your relevant debug logs

```text
Logger: homeassistant.components.websocket_api.http.connection
Kilde: components/websocket_api/commands.py:241
integration: Home Assistant WebSocket API (dokumentation, problemer)
Første forekomst: 10.35.44 (44 forekomster)
Senest logget: 10.39.14

[547383337328] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 241, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2741, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2784, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/services.py", line 452, in set_maximum_feed_grid_power
    bridge, uc = get_inverter_bridge(hass, service_call)
    ^^^^^^^^^^
TypeError: cannot unpack non-iterable HuaweiSolarBridge object
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

closed 2024-06-16 by wlcrs

wlcrs, 2024-06-16:
Whoops, sorry about that. I've released v1.4.1 with the fix. 

I've added an extra sensor which allows you to see the current active power control setting to make it up to you ;-) You need to enable it, as it is disabled by default. You'll find it in the Diagnostics-section of your Inverter device.

Thomsen-79, 2024-06-16:
That worked👍🏼

mynkow, 2025-07-03:
@Thomsen-79 @wlcrs I have times where the active control becomes unavailable. All other stats are working but I am out of control. This happens for 2 inverters connected to each other

commit 617de339 (2024-06-16): Fix active power control services. Add Active power control sensor.

Fixes #736