wlcrs/huawei_solar #737: [Bug]: Service calls incompatible with Retry (https://github.com/amitfin/retry)

acejoh, 2024-06-16:
### Describe the issue

I'm using the Retry component (https://github.com/amitfin/retry) to make sure my automations run correctly. This is especially important for my PV integration as I'm connecting to the inverter over wifi which sometimes causes actions to fail due to poor connection.
I'm not sure if this is an issue for you or for the retry component, I decided to ask here first as maybe the issue is not related to retry at all.

Some of the service calls that do not work with retry include:

- Huawei Solar: Forcible Charge/Discharge
- Huawei Solar: Set Active Power Control to Maximum Feed Grid Power
- Huawei Solar: Limit the power fed to the grid to percentage

### Describe your Huawei Solar Setup

Inverter Type: SUN2000-3KTL-L1
Inverter Firmware version: V200R001C00SPC139
sDongle present: No
sDongle Type: -
sDongle Connectivitiy: -
sDongle Firmware: -
Power meter present: single phase
Optimizers Present: No
Battery: LUNA2000-SO 5kWh
Battery Firmware version: V200R001C00SPC139
Huawei Solar integration version: 24.5.101.003

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[config_entry-huawei_solar-1feab6475f2a778adf423874db811461 (1).json](https://github.com/user-attachments/files/15857835/config_entry-huawei_solar-1feab6475f2a778adf423874db811461.1.json)

### Upload your relevant debug logs

```text
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 433, in validate_mapping
raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: extra keys not allowed @ data['entity_id']
2024-06-16 14:12:16.028 DEBUG (MainThread) [pymodbus.logging] send: 0xec 0x14 0x0 0x0 0x0 0x6 0x0 0x6 0xc3 0x4f 0x0 0x1
2024-06-16 14:12:16.030 DEBUG (MainThread) [pymodbus.logging] Adding transaction 60436
2024-06-16 14:12:16.031 ERROR (MainThread) [custom_components.retry] [Failed]: attempt 7/7: huawei_solar.set_maximum_feed_grid_power_percent(entity_id=sensor.inverter_day_active_power_peak, power_percentage=0)
Traceback (most recent call last):
File "/config/custom_components/retry/__init__.py", line 404, in async_retry
await self._hass.services.async_call(
File "/usr/src/homeassistant/homeassistant/core.py", line 2284, in async_call
processed_data: dict[str, Any] = handler.schema(service_data)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 272, in __call__
return self._compiled([], data)
^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 595, in validate_dict
return base_validate(path, iteritems(data), out)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 433, in validate_mapping
raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: extra keys not allowed @ data['entity_id']
2024-06-16 14:12:16.052 ERROR (MainThread) [custom_components.retry] [Failed]: attempt 7/7: huawei_solar.set_maximum_feed_grid_power_percent(entity_id=sensor.inverter_total_yield, power_percentage=0)
Traceback (most recent call last):
File "/config/custom_components/retry/__init__.py", line 404, in async_retry
await self._hass.services.async_call(
File "/usr/src/homeassistant/homeassistant/core.py", line 2284, in async_call
processed_data: dict[str, Any] = handler.schema(service_data)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 272, in __call__
return self._compiled([], data)
^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 595, in validate_dict
[... 74 lines cut ...]
2024-06-16 14:12:16.195 ERROR (MainThread) [custom_components.retry] [Failed]: attempt 7/7: huawei_solar.set_maximum_feed_grid_power_percent(entity_id=sensor.inverter_input_power, power_percentage=0)
Traceback (most recent call last):
File "/config/custom_components/retry/__init__.py", line 404, in async_retry
await self._hass.services.async_call(
File "/usr/src/homeassistant/homeassistant/core.py", line 2284, in async_call
processed_data: dict[str, Any] = handler.schema(service_data)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 272, in __call__
return self._compiled([], data)
^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 595, in validate_dict
return base_validate(path, iteritems(data), out)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 433, in validate_mapping
raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: extra keys not allowed @ data['entity_id']
```

### Please confirm the following:

- [ ] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

gieljnssns, 2024-06-16:
Can you post your automation?

dkmedic, 2024-06-16:
Same here not able to limit power to grid

LarsLautrup, 2024-06-16:
> Same here not able to limit power to grid

Same issue here. Can't reset the limit either.

dkmedic, 2024-06-16:
I have download (redownloaded) version 1.4.1 that solved the problem

acejoh, 2024-06-16:
> Can you post your automation?
Sure! Here's an example:

```
alias: Feed to grid when prices above 0
description: |-
  Allow inverter to feed to grid
trigger:
  - platform: numeric_state
    entity_id:
      - sensor.current_electricity_market_price
    above: -0.001
condition: []
action:
  - service: retry.actions
    metadata: {}
    data:
      sequence:
        - service: huawei_solar.reset_maximum_feed_grid_power
          data:
            device_id: 68036f75d87d567b7f73f4d7cd9324e7
    enabled: true
  - service: huawei_solar.reset_maximum_feed_grid_power
    metadata: {}
    data:
      device_id: 68036f75d87d567b7f73f4d7cd9324e7
    enabled: false
mode: single
```

This was generated by using the visual editor to define the Retry actions.
I just updated to 1.4.1 and will try again tomorrow!

Kind regards

acejoh, 2024-06-19:
> ```
> alias: Feed to grid when prices above 0
> description: |-
>   Allow inverter to feed to grid
> trigger:
>   - platform: numeric_state
>     entity_id:
>       - sensor.current_electricity_market_price [...]

I tried again after updating to 1.4.1 & restarting HA but the error persists:
This error originated from a custom integration.

Logger: custom_components.retry
Source: custom_components/retry/__init__.py:406
integration: Retry (documentation, issues)
First occurred: 10:34:35 PM (84 occurrences)
Last logged: 10:35:38 PM

[Failed]: attempt 7/7: huawei_solar.reset_maximum_feed_grid_power(entity_id=sensor.inverter_total_dc_input_energy)
[Failed]: attempt 7/7: huawei_solar.reset_maximum_feed_grid_power(entity_id=sensor.inverter_total_yield)
[Failed]: attempt 7/7: huawei_solar.reset_maximum_feed_grid_power(entity_id=sensor.inverter_pv_2_current)
[Failed]: attempt 7/7: huawei_solar.reset_maximum_feed_grid_power(entity_id=sensor.inverter_efficiency)
[Failed]: attempt 7/7: huawei_solar.reset_maximum_feed_grid_power(entity_id=sensor.inverter_pv_1_voltage)
Traceback (most recent call last):
  File "/config/custom_components/retry/__init__.py", line 406, in async_retry
    await self._hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 2284, in async_call
    processed_data: dict[str, Any] = handler.schema(service_data)
                                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 595, in validate_dict
    return base_validate(path, iteritems(data), out)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 433, in validate_mapping
    raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: extra keys not allowed @ data['entity_id']

closed 2024-07-30 by github-actions[bot]