wlcrs/huawei_solar #738: [Feature Request]: voltage control

home409ca, 2024-06-16:
### Describe your feature request

Ten minute overpressure protection point (42290)
Power on voltage upper limit (42084)
Upper limit of power grid reconnection voltage (42088)

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

Roving-Ronin, 2024-06-18:
@home409ca I know this is for Thijs (WLCRS) but reading out of interest, but left wondering what your actually wanting (and is this something we can already do via automations + existing services) ?

Any chance you could fully explain what your asking Thijs for, not just bullet points?

TIA.

home409ca, 2024-06-18:
@Roving-Ronin 

I would like to control these three parameters.

By default, in Poland the maximum voltage when the inverter can work is 253V.

Sometimes, I would like to use a higher voltage so that the inverter does not turn off (for some time), and when the voltage in the network is below 253V, I would like to restore the previous values ​​of these three parameters.

__
I cannot have a permanently entered higher voltage, because the requirements in Poland do not allow it and I can get a penalty.

Roving-Ronin, 2024-06-19:
@home409ca 

Up to Thijs as the owner, but that sounds like wanting to dynamically change the 'Grid Codes', that set those and a bunch of other settings that regulate the operation of the inverter.

If your anything like other countries the grid codes are mandated by the electricity network service providers (& their respective governments) and operating outside of the grid codes for your country / power network is illegal at worst, otherwise likely to to affect your warranty at the least.

Too high a voltage in the power network is something that also doesn't just potentially affect your own inverter/home, the over voltage actually also can destabilise the grid and cause outages/damage. This is why in some countries they either limit the kW your system can export, remotely turn off your whole inverter (yuck) or in newer instances are moving to Dynamic Export Control (DeX) that allows the DNSP to dynamically throttle up and down a houses inverter(s) to meet demand on the grid and not see over voltages on local grids. (Now if only we could get the info on accessing the API for this and feed into HA and have it dynamically control the Huawei / iStore inverters).

home409ca, 2024-06-19:
I don't want to change the network codes, but only these 3 parameters. Currently I do it manually using the FusionSolar app on my phone, but it's cumbersome and I can't control it from the HA level.

I know the risks and I know what too high voltage does. However, in my country, the upper voltage limit is 253V, even though in nearby countries this limit is even 264/270V, and electrical devices are the same and don't burn :)

[image]

That's why I'm asking for the possibility of control.

wlcrs, 2024-07-21:
I'm not going to add code to help people evade laws. I'm even less inclined to add sensors that are potentially dangerous when manipulated by ill informed users.

closed 2024-07-21 by wlcrs