wlcrs/huawei_solar #740: [Bug]: Battery sensors not reloaded after update to 1.4.1

stumpf84, 2024-06-18:
### Describe the issue

After updating to from v1.3.3 to v1.4.1 there is a new entity 'Battery 1' and the values for the old and new battery entity is only loaded on startup of the integration. 
The polling frequency is set to 10 seconds

```
alias: Huawei Solar inverter data polling
description: ""
trigger:
  - platform: time_pattern
    seconds: /10
condition: []
action:
  - service: homeassistant.update_entity
    data: {}
    target:
      entity_id: sensor.inverter_tagesertrag
mode: single

```

ModbusProxy is used.
All values except the battery entities are correctly updated.
Found this in the debug logging:

> ValueError: invalid literal for int() with base 10: 'discharging'
> ValueError: could not convert string to float: 'discharging'

### Describe your Huawei Solar Setup

Inverter Type: SUN2000-10KTL-M1
Inverter Firmware version: V100R001C00SPC159
sDongle present: Yes
sDongle Type: sDongleA-05 (WiFi / Ethernet)
sDongle Connectivitiy: Ethernet
sDongle Firmware: V200R022C10SPC108
Power meter present: three phase
Optimizers Present: No
Battery: LUNA2000-SO 10kWh
Battery Firmware version: V100R002C00SPC127
Huawei Solar integration version: 1.4.1

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[config_entry-huawei_solar-188103012c9738d35d122f30b29c5b16.json](https://github.com/user-attachments/files/15882212/config_entry-huawei_solar-188103012c9738d35d122f30b29c5b16.json)

### Upload your relevant debug logs

```text
2024-06-18 09:31:47.791 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-06-18 09:31:47.792 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-06-18 09:31:49.973 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' accesses hass.components.frontend. This is deprecated and will stop working in Home Assistant 2024.9, it should be updated to import functions used from frontend directly at custom_components/hacs/frontend.py, line 68: hass.components.frontend.async_register_built_in_panel(, please create a bug report at https://github.com/hacs/integration/issues
2024-06-18 09:31:50.863 ERROR (MainThread) [homeassistant.components.sensor] Error adding entity sensor.vog_l29_akkustatus for domain sensor with platform mobile_app
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 658, in state
    numerical_value = int(value)
                      ^^^^^^^^^^
ValueError: invalid literal for int() with base 10: 'discharging'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 661, in state
    numerical_value = float(value)
                      ^^^^^^^^^^^^
[... 243 lines cut ...]
2024-06-18 09:39:46.566 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x67 0x0 0x0 0x0 0x6b 0x1 0x3 0x68 0x0 0x0 0x4 0x8c 0xf 0xb0 0xf 0xb6 0xf 0x83 0x9 0xe 0x9 0x32 0x8 0xd3 0x0 0x0 0x0 0xf8 0x0 0x0 0x0 0xfe 0x0 0x0 0x1 0xc 0x0 0x0 0x3 0xe3 0x0 0x0 0x0 0x72 0xff 0xff 0xff 0xff 0x3 0xe8 0x13 0x8a 0x27 0x10 0x1 0xbb 0xb 0xb8 0x2 0x0 0x0 0x0 0x66 0x71 0x1c 0x48 0xff 0xff 0xff 0xff 0x0 0x0 0x0 0x60 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0xc0 0x0 0x6 0x40 0x63 0x0 0x7 0x2b 0x8e 0x66 0x71 0x55 0xd0 0x0 0x0 0x0 0x4 0x0 0x0 0x0 0xf5 old_data:  addr=None
2024-06-18 09:39:46.567 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x67 0x0 0x0 0x0 0x6b 0x1 0x3 0x68 0x0 0x0 0x4 0x8c 0xf 0xb0 0xf 0xb6 0xf 0x83 0x9 0xe 0x9 0x32 0x8 0xd3 0x0 0x0 0x0 0xf8 0x0 0x0 0x0 0xfe 0x0 0x0 0x1 0xc 0x0 0x0 0x3 0xe3 0x0 0x0 0x0 0x72 0xff 0xff 0xff 0xff 0x3 0xe8 0x13 0x8a 0x27 0x10 0x1 0xbb 0xb 0xb8 0x2 0x0 0x0 0x0 0x66 0x71 0x1c 0x48 0xff 0xff 0xff 0xff 0x0 0x0 0x0 0x60 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0xc0 0x0 0x6 0x40 0x63 0x0 0x7 0x2b 0x8e 0x66 0x71 0x55 0xd0 0x0 0x0 0x0 0x4 0x0 0x0 0x0 0xf5
2024-06-18 09:39:46.567 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2024-06-18 09:39:46.567 DEBUG (MainThread) [pymodbus.logging] Getting transaction 103
2024-06-18 09:39:46.567 DEBUG (MainThread) [pymodbus.logging] Frame check, no more data!
2024-06-18 09:39:46.567 DEBUG (MainThread) [pymodbus.logging] [0, 1164, 4016, 4022, 3971, 2318, 2354, 2259, 0, 248, 0, 254, 0, 268, 0, 995, 0, 114, 65535, 65535, 1000, 5002, 10000, 443, 3000, 512, 0, 26225, 7240, 65535, 65535, 0, 96, 0, 0, 0, 0, 0, 0, 0, 0, 448, 6, 16483, 7, 11150, 26225, 21968, 0, 4, 0, 245]
2024-06-18 09:39:46.567 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x04\x8c']
2024-06-18 09:39:46.567 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x03\xe3']
2024-06-18 09:39:46.567 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00r']
2024-06-18 09:39:46.568 DEBUG (MainThread) [pymodbus.logging] handle: [b'fq', b'\x1cH']
2024-06-18 09:39:46.568 DEBUG (MainThread) [pymodbus.logging] handle: [b'\xff\xff', b'\xff\xff']
2024-06-18 09:39:46.568 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x06', b'@c']
2024-06-18 09:39:46.568 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x07', b'+\x8e']
2024-06-18 09:39:46.568 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x04']
2024-06-18 09:39:46.568 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\xf5']
2024-06-18 09:39:46.568 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching BT2330398204_inverter_data_update_coordinator data in 3.771 seconds (success: True)
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

celodnb, 2024-06-18:
I'm seeing the exact same issue, for now I'm staying with the older version of the integration...

stumpf84, 2024-06-18:
> I'm seeing the exact same issue, for now I'm staying with the older version of the integration...

Maybe a noob question but how can I downgrade to 1.3.3?
When selecting the version by clicking 'Redownload' the version always resets to 1.4.1.
[image]

The 'Learn more' link leads to a non existing site.

celodnb, 2024-06-18:
Maybe someone else should answer, that has more knowledge than me, but I also couldn't seem to downgrade the integration, so I ended up rolling HA back to my last backup (I make a backup reach morning, so I had one from a few hours before), where the integration was on 1.3.3.

Until someone can help explain why the battery entities don't update, or what I need to do to fix it (of anything at all), or explain how it now works, I won't be upgrading this integration.

stumpf84, 2024-06-18:
This is the correct link for the downgrade documentation
[https://experimental.hacs.xyz/docs/use/entities/update/](url)

Downgrading to 1.3.3 seems to work.
Since the battery is fully charged and the sun is shining, I cannot check the SOC and Charge/Discharge power but the Bus voltage is updated periodically.

Roving-Ronin, 2024-07-21:
When updating to v1.4.x be aware that there was major changes to the code that saw 'Batteries' added, with this providing sensors that reflect the values of ALL physical batteries (ie if you have 2 x LUNA ESS) attached to Inverter 1.
Additionally some sensors were changed and/or removed from Battery 1/2 etc.
IMHO I would suggest deleting the plant (setup) you have in the Huawei Integration, then adding it back in from scratch. This will ensure you have any legacy sensors removed and all the sensors correctly named, as iirc there was also a name format change to the sensors going to v1.4.x

wlcrs, 2024-07-21:
To provide more flexibility in how the polling is done, inverter, power meter and battery sensors are not linked together anymore for updates. You must configure separate update triggers for all of them, as explained in https://github.com/wlcrs/huawei_solar?tab=readme-ov-file#inverter-polling-frequency

closed 2024-07-21 by wlcrs