wlcrs/huawei_solar #741: [Bug]: v1.4 - Observations

Roving-Ronin, 2024-06-18:
### Describe the issue

Hi Thijs,

Just updated a HA instance from 1.4.0a3 to v1.4 (and now 1.4.1) and noticed:

1) After updating and restarting that the Huawei / WLCRS integration had disabled itself. Integration worked fine after realising what had happened and re-enabling it. Just mentioning this, in case others see the same situation occur. 
On the flipside, updated my own HA ihstance (and updated to 2024.6.3) and rebooted and didn't have this happen.

2) Looking at remote site with a battery and noticed there is a 'Sensor: Battery 1 Remaining charge/discharge time'.  Is this a new sensor, as I don't recollect seeing it before?

Also I not that even though the battery is currently charging from PV, this sensor is showing '0 min'.
Edit: 36 hours later and it still hasn't updated from 0 min.

[image]

[image]

3) Looking at the unique ID for the 'Battery 1 Remaining charge/discharge time' it is listed as "sensor.battery_1_none_2". Wondering if this is correct or a typo with the 'none' (i.e. forgot to update the formula to name it something like "sensor.battery_1_remaining_time_2" for instance) ?  Assume this typo flows through to sensors for additional batteries?

4) In v1.4.1 the new Inverter 'Active Power Control' that reports the limit (or unlimited) that the inverter is set too. Noticed that this sensor is appearing in all inverters in a cascade. Given this sensor obtains it data from the primary inverter (as it manages the overall setting for the combined value of all inverters managed by the primary inverter) is there a need for having this sensor made available upon all but the first / primary inverter? (Checked two sites with dual inverters, activated this sensor on the seconday inverter is it remains in unknown state fyi).  Assuming its possible to filter/block it from all but the primary inverter, is it worth deleting it for all expect the primary inverter?

Thanks for the great work and updates Thijs !

### Describe your Huawei Solar Setup

Inverter Type: 2 x L1 5kW inverters cascaded. Currently only primary inverter is operating (as it has a LUNA ESS attached)
Inverter Firmware version: V200R001C00SPC138
sDongle present: Yes
sDongle Type: sDongleA-05
sDongle Connectivitiy: WiFi
sDongle Firmware: V200R022C10SPC108
Power meter present: Single Phase
Optimizers Present: No
Battery: LUNA2000-SO 10kWh
Battery Firmware version: V100R002C00SPC125
Huawei Solar integration version: v1.4.1

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[config_entry-huawei_solar-e7190a773e4143ed923016cd2d3a7e21.json](https://github.com/user-attachments/files/15882260/config_entry-huawei_solar-e7190a773e4143ed923016cd2d3a7e21.json)

### Upload your relevant debug logs

```text
Trying to drag and drop and fails.
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

amit-nz, 2024-06-20:
Just wanted to add that I have the same issue - 0 minutes remaining for charge/discharge time and mdi icon is missing also.

Roving-Ronin, 2024-06-26:
It seems that heinemannj is also seeing this 'Remaining charge/discharge time' remains at '0 min' issue. See in this thread:  https://github.com/wlcrs/huawei_solar/issues/746#issuecomment-2189303477

wlcrs, 2024-07-21:
2. Somebody suggested that this integration only makes sense in installations with a backup battery box?
3. Fixed with commit above
4. This is not documented by Huawei. Limiting it to the primary inverter would require me to write extra code. Is there any harm in having that entity available in the secondary inverters?

closed 2024-07-21 by wlcrs

Roving-Ronin, 2024-07-21:
Re 2. Just jumped onto a site I have access to with 2 x 5kW L1 and 1 x 15kW LUNA S0, including a Backup-Box (1p) and as usual it always APPEARS to be showing '0 min'. However, changing to show the last month of history its actually getting some data:

[image]

What exactly is this sensor supposed to reflect / mean, as from the name you'd think it would be constantly updating to show how long to charge it or discharge during 'normal' operation.

Re 4. Not in the documentation, but that is how the inverters work, with the primary taking the overall limit and applying it across all the inverters in the cascase. On the above 2 x 5kW L1 cascade for example, it shows:

[image]

Whilst the secondary inverter for this sensor remains in 'unknown' state (as the primary inverter is controlling this for the whole cascade of inverters).

BTW this raises the question of is it worth looking to add an 'Inverters' (same as 'Batteries' was added in v.1.4.x), to contain sensors such as this? This might also reduce additional code to 'filter' it to only the primary inverter in a cascade, as instead it wouldn't be added to type=inverter, it would be for inverter(s), of which there would only be one as Huawei only supports up to 3 inverters in a cascade for a plant. I believe there would be a bit of coding required to do this, so maybe a 'longer term' goal to work towards?

commit 10e32016 (2024-07-21): Fix icon of sensor

Fixes #741