wlcrs/huawei_solar #745: [Feature Request]: Register: 37007 - batteries_rated_charge_power

heinemannj, 2024-06-22:
### Describe your feature request

`batteries_rated_charge_power`

Register: `37007`

**Really helpful for energy management optimazations**

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

Roving-Ronin, 2024-06-24:
@heinemannj 

An observation, the Huawei Modbus Registers Definitions (Issue 05 - 2023-02-16 is the last I've found) list this register, being for "[Energy storage unit 1] Rated charge power".

However there is no register defined for "[Energy storage unit **2**] Rated charge power"  (i.e. if a user has 2 [or more] batteries installed).

heinemannj, 2024-07-13:
Must be a **wrong description in Huawei Modbus Registers documentation**:

[image]
[image]

**`Max output power for a LUNA2000 can never be greater than 5kWh!`**

My setup contains:

|LUNA2000|Max output power|
|------------------------|---------|
|1 x LUNA2000-5-S0   | 2.5 kWh|
|1 x LUNA2000-10-S0 | 5.0 kWh|
|Total                           | 7.5 kWh|

Roving-Ronin, 2024-07-13:
37007 should only report 5000W, maximum of it's a 10/15kWh LUNA. This being as that register is for Battery #1 (only).

37046 would be the better register to query as it's for 'Batteries', so would show the capacity for all attached batteries combined (ie 5000 + 2500 = 7500W, as your setup reports).

Just wondering as to use case for this, won't the Maximum Charge (number) under Batteries that v1.4 added do the same thing, or you want to see maximum possible versus the maximum that has been defined (in the number input)?

PS. iirc New LUNA S1 (7/14/21kWh) at its maximum goes up to 10kWh dis/charge.

heinemannj, 2024-07-13:
**My final solution**: https://github.com/wlcrs/huawei_solar/issues/746

wlcrs, 2024-07-21:
Done in #768

closed 2024-07-21 by wlcrs