wlcrs/huawei_solar #746: [Feature Request]: Register: 37009 - batteries_rated_discharge_power

heinemannj, 2024-06-22:
### Describe your feature request

`batteries_rated_discharge_power`

Register: `37009`

**Really helpful for energy management optimazations**

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

Roving-Ronin, 2024-06-24:
@heinemannj 

An observation, the Huawei Modbus Registers Definitions (Issue 05 - 2023-02-16 is the last I've found) list this register, being for "[Energy storage unit 1] Rated discharge power".

However there is no register defined for "[Energy storage unit 2] Rated discharge power"  (i.e. if a user has 2 [or more] batteries installed).

heinemannj, 2024-06-25:
I've installed a second Luna (5kWh) since a couple of days.

After adding the second one `batteries_rated_discharge_power` have been increased from 5000W to 7500W.
Clearly visible in FusionSolar portal.

Roving-Ronin, 2024-06-25:
@heinemannj I assume as in Battery1 is a 10kWh or 15kWh providing 5000W maximum charge/discharge capacity, and Battery2 at 5kWh with 2500W maximum charge/discharge capacity, hence it showing as 7500W.

As I posted above, Huawei documentation has this listed as "[Energy storage unit 1] Rated discharge power", with this name formatting used to refer to physical battery 1. From what your seeing though, it would seem the Huawei documentation is incorrect and it should read as "[Energy storage] Rated discharge power" (as Huawei refers to other sensors that are for the 'overall' battery setup, or what WLCRS v1.4 recently added with the "Batteries" sensors).

heinemannj, 2024-06-25:
**My actual inverter setup** as visible in FusionSolar portal:

[image]
[image]

**My actual batteries setup** as visible in FusionSolar portal (in **summary: 10kWh + 5kWh**):
[image]

LUNA2000-1:
[image]

LUNA2000-2:
[image]

Device infos **by HA Huawei Solar Integration**:
[image]

[image]

**Interesting notes**:

- `number.inverter_power_derating`
- `number.batteries_grid_charge_maximum_power`
- `number.batteries_maximum_discharging_power`

has all the **proper max values of 6600W**.

- ` number.batteries_maximum_charging_power`

has the **proper max value of 7500W**.

**Conclusion**:

In HA Huawei Solar Integration **somewhere in place but not distributed via sensors**:

- inverter_maximum_active_power: 6600
- batteries_rated_discharge_power: 6600
- batteries_rated_charge_power: 7500

In HA Huawei Solar Integration **not in place and not distributed via sensors**:

- inverter_rated_power: 6000
- batteries_rated_capacity: 15
- batteries_rated_charge_energy: 17.250

In addition some discrepancies in LUNA and LUNA 2 devices:

[image]

[image]

I've found an **easy workarround** e.g.:

```
{%- macro globalVariables() %}
    {{- {
            'inverter_rated_power': 6000,
            'inverter_maximum_active_power': 6600,
            'batteries_rated_capacity': 15,
            'batteries_rated_charge_energy': 17.250,
            'batteries_rated_discharge_power': 6600,
            'batteries_rated_charge_power': 7500,
            'batteries_charge_end_time_expected': '14:00',
            'batteries_soc_setpoint_low': 15,
            'epexspot_quantile_setpoint_low': 0.15,
            'epexspot_quantile_setpoint_neutral': 0.33,
            'epexspot_quantile_setpoint_high': 0.8
        }|to_json -}}
{%- endmacro %}
```

```
{% from '032-emhass.jinja' import globalVariables %}
{% set var = globalVariables()|from_json %}
{% set bat_state = states('sensor.batteries_state_of_capacity')|float(0) / 100 %}
{{ (var.batteries_rated_capacity * bat_state) }}
```

**But this workarround is far away to maintain blueprints for millions of people to safe our earth ...**

**My vision**:

[image]

[image]

[image]

Roving-Ronin, 2024-06-27:
@heinemannj 
Sorry where's the 'easy workaround' actually installed and where is it scraping its data from?
Also, would you please be able to get back to me in regards working with others to standardise the derived sensors, entity names/ID's etc that we are all using? 
I'd love to be able to leverage your dashboards also, and also use Nick Jensens' "PEES" package for tracking its data, and SunSynk Card but at the moment each uses different entities / names / IDs etc  so its a nightmare to use more than one. Working to standardise upon a  'derived sensors standardised package' that all Huawei (& iStore in Australia) users could use going forward seems logical. ??

heinemannj, 2024-06-27:
https://www.home-assistant.io/docs/configuration/templating/#reusing-templates

[image]

You only have to create the template file `032-emhass.jinja` in the folder `config/custom_templates` with the following content:

```
{%- macro globalVariables() %}
    {{- {
            'inverter_rated_power': 6000,
            'inverter_maximum_active_power': 6600,
            'batteries_rated_capacity': 15,
            'batteries_rated_charge_energy': 17.250,
            'batteries_rated_discharge_power': 6600,
            'batteries_rated_charge_power': 7500,
            'batteries_charge_end_time_expected': '14:00',
            'batteries_soc_setpoint_low': 15,
            'epexspot_quantile_setpoint_low': 0.15,
            'epexspot_quantile_setpoint_neutral': 0.33,
            'epexspot_quantile_setpoint_high': 0.8
        }|to_json -}}
{%- endmacro %}
```
Templates in this folder will be loaded at startup. To reload the templates without restarting Home Assistant, invoke the [homeassistant.reload_custom_templates](https://my.home-assistant.io/redirect/developer_call_service?service=homeassistant.reload_custom_templates) service.

Roving-Ronin, 2024-06-27:
@heinemannj 

Hmm... OK. Yes that really isn't sustainable long term, as you commented, and not 'dynamic' for updating to what someone has as their setup (given it looks to be static defined values).
Looking at my /custom_templates directory and it only has a personal.jinja and power_control.jinja file. Looking at the name of the file you reference, it's actually another developers file also (EMHASS). There are some workarounds that I've been looking at for SunSynk and PEES (to automate setup), but really need Thijs to agreed to adding the additional sensors. Hopefully this would be entertained with the architecture changed in v1.4.x  (and set to disabled by default).

BTW could you please comment re the standardisation of sensors package work group...? Logic would dictate this as a common sense direction, through which most will benefit vs continuing with duplicating / maintaining 3-4 seperate sets of sensors. Thanks.

heinemannj, 2024-06-27:
Standardisation of sensors package work groups is out of topic for this request!

My first view: Naming convention of PEES at this time is not fitting to my actual settings ...
My approach: To be much close to the nameing conventions provided by @wlcrs huawei_solar and all other integrations I needs to synchronize:

- huawei_solar
- epexspot
- tibber
- sun
- sun2
- solcast
- emhass
- myVaillant
- dwd_DeutscherWetterdienst
- Shelly
- ...

heinemannj, 2024-07-13:
**My final solution and proposal:**

**Remove from Battery 1** as the `LUNA200 max output power is limited to 5kWh`:

[image]

**Add to Batteries**:

[image]

My setup contains below and is fitting exactly above figures:

|LUNA2000|Max output power|Capacity|
|------------------------|---------|--------|
|1 x LUNA2000-5-S0   | 2.5 kWh|5 kWh|
|1 x LUNA2000-10-S0 | 5.0 kWh|10 kWh|
|Total                           | 7.5 kWh|15 kWh|

Inverter: `SUN2000-6KTL-M1 (High Current)`

**Required code changes**:

```
    HuaweiSolarSensorEntityDescription(
        key=rn.STORAGE_RATED_CAPACITY,
        icon="mdi:home-battery",
        native_unit_of_measurement=UnitOfEnergy.WATT_HOUR,
        state_class=SensorStateClass.MEASUREMENT,
        entity_category=EntityCategory.DIAGNOSTIC,
    ),
    HuaweiSolarSensorEntityDescription(
        key=rn.STORAGE_MAXIMUM_CHARGE_POWER,
        icon="mdi:home-battery-outline",
        native_unit_of_measurement=UnitOfPower.WATT,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.POWER,
        entity_category=EntityCategory.DIAGNOSTIC,
    ),
    HuaweiSolarSensorEntityDescription(
        key=rn.STORAGE_MAXIMUM_DISCHARGE_POWER,
        icon="mdi:home-battery-outline",
        native_unit_of_measurement=UnitOfPower.WATT,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.POWER,
        entity_category=EntityCategory.DIAGNOSTIC,
    ),
```
```
            "storage_rated_capacity": {
                "name": "Rated Capacity"
            },
            "storage_maximum_charge_power": {
                "name": "Maximum charge power"
            },
            "storage_maximum_discharge_power": {
                "name": "Maximum discharge power"
            },
```

Roving-Ronin, 2024-07-14:
@heinemannj 
I've created a branch in my fork, ready to request a PULL into WLCR integration at: https://github.com/Roving-Ronin/istore_solar/tree/Energy-Management-Optimisation-Sensors (including all the language files).

I'm not sure about the removing of the sensors from the individual batteries though. Could these still be of use to people? Would be good if Thijs would provide his views? (If possible, then will create a PULL based upon that feedback).

As an aside, whilst editing this and looking at entries such as:
- Active power Control (existing. If operated multiple inverters this represents the value for ALL inverters)
- Active Power (exists for each individual inverter, no 'Inverters' combined value)
- Daily Yield (exists for each individual inverter, no 'Inverters' combined value)
- Total Yield (exists for each individual inverter, no 'Inverters' combined value)

is that like how v1.4 introduced the 'Batteries' device, it would be really good if an 'Inverters' device could be added to the integration.

PS. In your code below (for use in en.json and strings.json) you have a typo. The P in P_max needs to be all lower case.
```
            "P_max": {
                "name": "Max active power"
            }
```

heinemannj, 2024-07-14:
> @heinemannj I've created a branch in my fork, ready to request a PULL into WLCR integration at: https://github.com/Roving-Ronin/istore_solar/tree/Energy-Management-Optimisation-Sensors (including [...]

Remove the misleeding sensors and make the pull request:
- They have wrong values
- Inserted into the wrong device
- Fitting solution introduced

In any case Thijs needs to approve - If he has some doubt by reviewing, he will ask and discuss ...

Roving-Ronin, 2024-07-14:
Ah OK, thanks for explaining, was getting a bit confused as the thread grew.

I will update the changes in the fork above, to remove those sensors and then submit a PULL. Will do tomorrow, as about to crash for the night.

PS. Do you have an idea on how an 'Inverters' (plural) could be added ?

heinemannj, 2024-07-14:
> PS. Do you have an idea on how an 'Inverters' (plural) could be added ?

No - in addition this will be the biggests change since the last couple of years - to be initiated by the owner

wlcrs, 2024-07-21:
Done in #768

closed 2024-07-21 by wlcrs