wlcrs/huawei_solar #747: [Feature Request]: Register: 30075 - inverter_maximum_active_power

heinemannj, 2024-06-22:
### Describe your feature request

`inverter_maximum_active_power`

Register: `30075`

**Really helpful for energy management optimazations**

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

BartBourgeois, 2024-07-09:
Was actually looking for the same feature, this would allow running my solar in off-grid mode in the summer. 
I currently use:
  huawei_solar.reset_maximum_feed_grid_power
  huawei_solar.set_maximum_feed_grid_power
With the downside that setting the maximum to any value disables the inverter to 0W.
Not perfect, but doable. 

Beeing able to set "Maximum Active Power" would allow (i think) for an automatisch that sets this value to match my current power usage.

BartBourgeois, 2024-07-09:
It seems 30075 is not the correct register,
https://www.debacher.de/wiki/Sun2000_Modbus_Register

Seems to be Read-Only, probably stating the Max Power of the inverter.

**Parameter Setting by Fixed Value**
The host delivers the active scheduling target value, which is an absolute value measured in kW. The adjustment precision is 0.1 kW. The maximum value is the reported Pmax, and the minimum value is 0.

**Operation:**
Change the value of Active power control (register 40118) to 2, and then change the value of Fixed active power derating (register 40120) to the target value (kW).

For more details, please see the brochure

https://support.huawei.com/enterprise/en/doc/EDOC1100113918?section=k002

After testing with the SUN2000 App, i was able to set the "Maximum Active Power" and "Maximum Apparent Power",
works right away and the inverter changes output instantly to the desired value
[image]

Plan B, learn how to code and figure out these libraries :-)

BartBourgeois, 2024-07-09:
Moving on.... so lucky this library is well writen and structured

Added these lines in "register_names"
```py
MAXIMUM_APPARENT_POWER = "maximum_apparent_power"
MAXIMUM_ACTIVE_POWER = "maximum_active_power"
```

Added these lines in the "registers.py"
```py
rn.MAXIMUM_APPARENT_POWER: U32Register("kVA", 1000, 42021,  writeable=True),
rn.MAXIMUM_ACTIVE_POWER: U32Register("kW", 1000, 42023,  writeable=True),    
```

Set both values in the inverter to something easily recognizable
[image]

Adjusted the "bridge_test" code
```py
import asyncio
import logging

import huawei_solar

from huawei_solar import HuaweiSolarBridge
from huawei_solar import register_names as rn

loop = asyncio.new_event_loop()

logging.basicConfig(level=logging.DEBUG)

async def test():
    bridge = await HuaweiSolarBridge.create("192.168.1.123", 502, 1)

    print(await bridge.client.get(rn.MAXIMUM_APPARENT_POWER))
    print(await bridge.client.get(rn.MAXIMUM_ACTIVE_POWER))

    await bridge.stop()

loop.run_until_complete(test())
```

Got the correct results :-)
[image]

Next step, try to modify.... fingers crossed i dont brick the inverter

Roving-Ronin, 2024-07-12:
@BartBourgeois 
> Was actually looking for the same feature, this would allow running my solar in off-grid mode in the summer. I currently use: huawei_solar.reset_maximum_feed_grid_power [...]

What are you setting this service to restrict too? If your not connected to the grid you should be setting this to 0W / 0% and the inverter(s) should then be reacting to only produce to meet the demands of your 'in house' load.

I've just been testing this with different xxxx Watts and Reset to Maximum and it works fine, with the inverters producing to meet the load and grid export being curtailed to the set maximum watts.

PS. You haven't commented upon your setup (what inverter, how many, consumption meter installed etc).
 
> Beeing able to set "Maximum Active Power" would allow (i think) for an automatisch that sets this value to match my current power usage.

If you set this up using 42021 / 42023 that restricts the actual production of the inverter to a maximum watts, versus using 30075 (Max Active Power), and if needed 30077 - Max Apparent Power, then you could see the situation where (lets assume you have a 10kWh M1 inverter) you set your 42021 to be able to produce a maximum of 5000W and if you turned on your AirCon/Oven etc and you needed 7500W, then inverter would only produce the maximum of 5000W, probably giving you an outage / breakers trip etc.

If you (sort out the issue your seeing and) use the 'service: huawei_solar.set_maximum_feed_grid_power' and had set that to 0W (or the % alternative version), then your inverter would be able to ramp up to produce the requested 7500W for 'in house' consumption, yet your exports to the grid should remain 0W.

BartBourgeois, 2024-07-12:
@Roving-Ronin ,

Thanks for the insights, might already explain why the settings does not work in my setup.
The PV-Inverter is not actually in Offgrid mode. My choice of words was not ideal. 

>If your not connected to the grid you should be setting this to 0W / 0% and the inverter(s) should then be reacting to only produce to meet the demands of your 'in house' load.

We'll, it's not connected to ""The grid"", but it's connected to an already existing grid (Micro-grid generated by ESS).
I now understand why the term are confusing, by "Offgrid" I Mean not connected to the mains power supply.

>PS. You haven't commented upon your setup (what inverter, how many, consumption meter installed etc).

I'll explain my current setup, without the PV-Inverter first, should explain why the inverter is not in offgrid mode.

My main power 'sources':
- Grid power / 1-Fase 230V / 63A
- ESS (Victron) / 1-Fase 230V / 63A / 60Kwh Battery
 
Both power-sources are connected to a Automatic-Transfer-Switch (ATS), the ATS prefers the ESS input, can fail over to Grid if ESS fails.
I can also tell the ATS to switch to grid via Victron/HA, within the ESS the "Grid" is configured as generator. (Winter)
Main switch board (all loads) are connected to the ATS, so all loads comes from the ESS or Grid.
Power comes directly from the grid, of in the ESS the (Micro)grid is generated by the Victron inverters from battery power.

Basically the setup is a gaint UPS for the entire house, that can pull any load my grid connection can, with enough battery-capacity for +4 days without power. 

The ESS is managed with Victron, consuption meters are connected to the ESS, all integrated with HomeAssistant.
- Grid meter
- PV-Meter on Grid-side
- PV-Meter on ESS-Side
- Meter on main-switch board (Loads)

The ESS has a Dynamic mode that allows charging from grid when the prices are low (or even negative).
The ATS setup is done because 100% offgrid is not possible in Belgium due to weater, although with Solar in combination with the ESS I get around 8 ~ 9 months of 100% self-suffciency.

>PS. You haven't commented upon your setup (what inverter, how many, consumption meter installed etc).

Now the PV-Inverters comes into play.
PV-Inverter = 1x SUN2000-5KTL-L1  (6800Wp panels connected)
It's the only Huawei item in the installation, it's connected to a Manual-Transfer-Switch (MTS), break-before-make.
The MTS can switch the Solar panels between Grid and ESS. Because of regulations I was not allowed to place the PV-Inverter in the main board, behind the ATS. Automatic switching of PV-Inverter is not allowed. 
The PV-Inverter is by default connected to the ESS side, but in emergencies or when the ESS requires maintenance I can switch it back to grid. 
In both cases, the PV-Inverter connects to an already existing managed power supply, or the main-grid, or the micro-grid generated by the ESS. 

> What are you setting this service to restrict too? If your not connected to the grid you should be setting this to 0W / 0% and the inverter(s) should then be reacting to only produce to meet the demands of your 'in house' load.

The PV-Inverter has no idea about loads, all loads are carried by the grid of by the ESS. The PV-Inverter 'assists'.
It's the Victron system that can supply every power demand and react on sudden peaks, it can also charge the battery when there is access PV power.

> If you set this up using 42021 / 42023 that restricts the actual production of the inverter to a maximum watts, versus using 30075 (Max Active Power), and if needed 30077 - Max Apparent Power, then [...]

This doesn't play in my situation, since the PV-Inverter only assists where possible and all loads are carried by the Victron.
I can switch the solar off any-time, does not influence my off-grid ESS system.
But I do understand your the concern here, in some situaties misconfiguring this could lead to chaos/outage/trips.
Never-the-less I do not see this as a reason for not including the  "42023 / Maximum Active Power" setting into the library/HA Integration. I Presume that anybody with the Modbus running on the PV-Inverter already has "Installer" access, and thus access to way more parameters that could be nefarious.

> If you (sort out the issue your seeing and) use the 'service: huawei_solar.set_maximum_feed_grid_power' and had set that to 0W (or the % alternative version), then your inverter would be able to [...]

I presume now, this mode is not working because
- PV-Inverter is not in off-grid mode, does not create it's own grid
- PV-Inverter does no have any meter connected
- PV-Inverter does not carry any load by itself, just feeds the ESS.
So the settings work, but not in the setup I have.

When in ESS (Offgrid mode) the grid is actually disconnected, no access power is returned. 
When the batteries are 100% full, and the PV-Inverter generates more than my current load, this currently don't work 100%.
Certainly if the PV-Inverter is running at 100%. The Victron system detected the over-capacity (knows the load and battary SOC),
the it's start fequency shifting which the PV-Inverter reacts to...but in full load not fast enough. In a couple of minutes I hit 52Hz and the PV-Inverter shuts down (as it should). The frequency shifting works greak for small adjustments, but not in this case.
Altough it kinda-works in normload load situaties, I also don't like the frequency shifting part, want the internal system running at a stable 230V/50Hz.

[... cut ...]
The I tried making changes to the HA-Integration, but not figured out how I can modify the main library.
[image]

My plan was first working this out in my installation as a Proof-Of-Concept, then try to PullRequest my changes in both the base-library as the HA-Integration. I would rather have the changes permanent in the proper place, then modify my local version.

Just trying to get smarter and in the process getting this PV-Inverter integrated 100% into the ESS.
Which seems just a  "42023 / Maximum Active Power"  read/write away :-D

Please feel free to comment on my setup or anything else.

heinemannj, 2024-07-13:
**My final solution** for my original request: https://github.com/wlcrs/huawei_solar/issues/746

wlcrs, 2024-07-21:
Done in #768

closed 2024-07-21 by wlcrs

kevinjanisch, 2025-04-23:
> Moving on.... so lucky this library is well writen and structured
> 
> Added these lines in "register_names"
> 
> MAXIMUM_APPARENT_POWER = "maximum_apparent_power"
> MAXIMUM_ACTIVE_POWER = "maximum_active_power"
> Added these lines in the "registers.py"
> 
> rn.MAXIMUM_APPARENT_POWER: U32Register("kVA", 1000, 42021,  writeable=True),
> rn.MAXIMUM_ACTIVE_POWER: U32Register("kW", 1000, 42023,  writeable=True),    
> Set both values in the inverter to something easily recognizable [image]
> 
> Adjusted the "bridge_test" code
> 
> import asyncio
> import logging
> 
> import huawei_solar
> 
> from huawei_solar import HuaweiSolarBridge
> from huawei_solar import register_names as rn
> 
> loop = asyncio.new_event_loop()
> 
> logging.basicConfig(level=logging.DEBUG)
> 
> async def test():
>     bridge = await HuaweiSolarBridge.create("192.168.1.123", 502, 1)
> 
>     print(await bridge.client.get(rn.MAXIMUM_APPARENT_POWER))
>     print(await bridge.client.get(rn.MAXIMUM_ACTIVE_POWER))
> 
>     await bridge.stop()
> 
> loop.run_until_complete(test())
> Got the correct results :-) [image]
> 
> Next step, try to modify.... fingers crossed i dont brick the inverter

Is this planned at some point?
I currently use [BartBourgeois](https://github.com/BartBourgeois) fork because of the reasons he already stated.

Bleialf, 2025-04-23:
> Was actually looking for the same feature, this would allow running my solar in off-grid mode in the summer. I currently use: huawei_solar.reset_maximum_feed_grid_power [...]

Is this planned at some point?
I currently use [BartBourgeois](https://github.com/BartBourgeois) fork because of the reasons he stated