wlcrs/huawei_solar #748: [Feature Request]: Register: 30073 - inverter_rated_power

heinemannj, 2024-06-22:
### Describe your feature request

`inverter_rated_power`

Register: `30073`

**Really helpful for energy management optimazations**

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

Roving-Ronin, 2024-06-24:
@heinemannj 

This request is the same as I currently have raised in discussion with Thijs. See: https://github.com/wlcrs/huawei_solar/discussions/677#discussioncomment-9827841

This link includes what appears to be the registers that support querying this data. Alas my coding skills don't reach into the python world and I can't work out how to add them and submit a pull to the Huawei-Library. Hopefully Thijs would agree to add them with v1.4, even if disabled by default?

BTW what are you trying to do, as there's other efforts going on to provide some of these type of sensors outside of WLCR integration (using data from it). Such as:

[image]

Note: The Batteries rated capacity is the one above I'm trying to get going. There are registers for each battery module that can be queried, but that would be a nightmare from the Huawei-Library side (meaning 3 register per battery to call, and 2 batteries per inverter ...and potentially 4 per with the new LUNA2000 S1 7/14/21kWh model).

I've tried to tag you previously (hard given Github removed messaging) to ask if you would be interested in collaborating on standardising the derived sensors for Huawei (& iStore, as they are OEM'ed in Australia) + WLCRS. Currently you've got a good range of sensors for various things, https://github.com/JensenNick has a good package (or two) also tracking power flows, costs / savings / ROI etc.  

[image]

With differing versions of sensors, with different ID's / names etc we however end up with confusion/issues when using another package/card that expects its own sensor entities.  Such as:

[image]

Thinking that logical idea would be to combine and every agree on a standard set of derived sensors that 'any and all' addons and cards etc could leverage. ???

heinemannj, 2024-07-13:
**My final solution**: https://github.com/wlcrs/huawei_solar/issues/746

wlcrs, 2024-07-21:
Done in #768

closed 2024-07-21 by wlcrs