wlcrs/huawei_solar #749: [Feature Request]: Register for 'batteries_rated_capacity'

heinemannj, 2024-06-22:
### Describe your feature request

Unfortunately I've not found in Huawei Solar Inverter Modbus Interface Definitions ...
but really helpful for energy management optimazations.

Seen in FusionSolar portal:

[image]

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

Roving-Ronin, 2024-06-24:
@heinemannj 

This request is the same as I currently have raised in discussion with Thijs. See: https://github.com/wlcrs/huawei_solar/discussions/677#discussioncomment-9827841

This link includes what appears to be the registers that support querying this data. Alas my coding skills don't reach into the python world and I can't work out how to add them and submit a pull to the Huawei-Library. Hopefully Thijs would agree to add them with v1.4, even if disabled by default?

BTW what are you trying to do, as there's other efforts going on to provide some of these type of sensors outside of WLCR integration (using data from it). Such as:

[image]

Note: The Batteries rated capacity is the one above I'm trying to get going. There are registers for each battery module that can be queried, but that would be a nightmare from the Huawei-Library side (meaning 3 register per battery to call, and 2 batteries per inverter ...and potentially 4 per with the new LUNA2000 S1 7/14/21kWh model).

I've tried to tag you previously (hard given Github removed messaging) to ask if you would be interested in collaborating on standardising the derived sensors for Huawei (& iStore, as they are OEM'ed in Australia) + WLCRS. Currently you've got a good range of sensors for various things, https://github.com/JensenNick has a good package (or two) also tracking power flows, costs / savings / ROI etc.  

[image]

With differing versions of sensors, with different ID's / names etc we however end up with confusion/issues when using another package/card that expects its own sensor entities.  Such as:

[image]

Thinking that logical idea would be to combine and every agree on a standard set of derived sensors that 'any and all' addons and cards etc could leverage. ???

Roving-Ronin, 2024-07-12:
@heinemannj 

This register is already in the huawel library:  rn.STORAGE_RATED_CAPACITY: U32Register("Wh", 1, 37758),

Prior to seeing if Thijs would allow adding this sensor into the integration, by lodging a PULL request, would you like to try these changes and see if they work for you also ? 

In sensor.py add the STORAGE_RATED_CAPACITY section, so it sits between STORAGE_STATE_OF_CAPACITY abd STORAGE_RUNNING_STATUS. Like this:

```
BATTERIES_SENSOR_DESCRIPTIONS: tuple[HuaweiSolarSensorEntityDescription, ...] = (
    HuaweiSolarSensorEntityDescription(
        key=rn.STORAGE_STATE_OF_CAPACITY,
        icon="mdi:home-battery",
        native_unit_of_measurement=PERCENTAGE,
        state_class=SensorStateClass.MEASUREMENT,
    ),
    HuaweiSolarSensorEntityDescription(
        key=rn.STORAGE_RATED_CAPACITY,
        icon="mdi:battery-charging",
        native_unit_of_measurement=UnitOfPower.WATT,
        state_class=SensorStateClass.MEASUREMENT,
        entity_category=EntityCategory.DIAGNOSTIC,
    ),
    HuaweiSolarSensorEntityDescription(
        key=rn.STORAGE_RUNNING_STATUS,
        entity_category=EntityCategory.DIAGNOSTIC,
    ),
```

Then in strings.json, same as above add the RATED_CAPACITY section, so it looks like:

```
            "storage_state_of_capacity": {
                "name": "State of capacity"
            },
            "storage_rated_capacity": {
                "name": "Rated Capacity"
            },
            "storage_running_status": {
                "name": "Status"
            },
```

Restart HA and then see if in the Batteries device, in its Diagnostic section you have:

[image]

Then open the sensor and make sure the name is displaying as 'Batteries Rated Capacity' and (click the settings/cog wheel) that the Entity ID = sensor.batteries_rated_capacity.

heinemannj, 2024-07-13:
You have to change the unit from `Watt` to `KILO_WATT_HOUR`:

```
    HuaweiSolarSensorEntityDescription(
        key=rn.STORAGE_RATED_CAPACITY,
        icon="mdi:home-battery",
        native_unit_of_measurement=UnitOfEnergy.KILO_WATT_HOUR,
        state_class=SensorStateClass.MEASUREMENT,
        entity_category=EntityCategory.DIAGNOSTIC,
    ),
```

In addition I had to modify the `en.json` as well.

My entity_ID was `sensor.batteries_none` - potentially due to the issue that I had not modified `en.json` at my first try.

Roving-Ronin, 2024-07-13:
@heinemannj The Huawei documentation refers to the output in Wh, but I found it displated as '15,000'  (for a 15kWh LUNA) when I first tried it, hence I listed the Unit of Measurement as 'Watt'. What do you get when you use 'KILO_WATT_HOUR' ? I would have thought that would see you get '15,000kWh' ?

Ahh thanks re the en.json I totally overlooked that and was wondering where that was set.

heinemannj, 2024-07-13:
[image]

heinemannj, 2024-07-13:
Changing from `KILO_WATT_HOUR` to `WATT_HOUR` finally results to:

[image]

[image]

```
    HuaweiSolarSensorEntityDescription(
        key=rn.STORAGE_RATED_CAPACITY,
        icon="mdi:home-battery",
        native_unit_of_measurement=UnitOfEnergy.WATT_HOUR,
        state_class=SensorStateClass.MEASUREMENT,
        entity_category=EntityCategory.DIAGNOSTIC,
    ),
```

heinemannj, 2024-07-13:
**My final solution**: https://github.com/wlcrs/huawei_solar/issues/746

wlcrs, 2024-07-21:
Done in #768

closed 2024-07-21 by wlcrs