wlcrs/huawei_solar #754: [Feature Request]: Metering on separate phases

strongiukas, 2024-06-27:
### Describe your feature request

In our country, the grid is calculating electricity on 3 separate phases. 
For example: Inverter is generating 3kW but if I switch on an 2kW appliance on one phase, then our grid will count as: 2kW to grid, but 1kW from grid.
So I can't get correct numbers how much I consumed and how much I sent to grid.
Maybe it is possible to change measurment?
Inverter: Sun2000-10KTL-M1
SmartMeter: DTSU666-H

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

Roving-Ronin, 2024-07-04:
@strongiukas 
This integration only publishes the modbus registers that Huawei solar devices make available. From what your saying your country does not provide 'Nett Metering', but charges per phase. Huawei doesn't provide any modbus register that tracks import / export on a per phase basis, instead they only provide the overall import / export for all the phases.

Your going to need to utilise the utility_meter (or the HACS integration 'Energy_Meter' if you want it to also calculate your costs/income) and the Integral helper (Riemann sum to convert from Watts to kWh also). However before doing that could you explain more how the consumption / export is tracked in your country?

Whilst it sounds like its done per phase and not Nett of all phases, is it tracked Nett within each phase ? 
i.e. On say Phase A, if you import 10kWh for the day, but also export 5kWh on that phase, are you billed for 5kWh overall? Or are they charging for 10kWh (say at $0.50 per kWh) and then seperately paying you for 5kWh (usually at a lesser amount like $0.05 per kWh) ?

strongiukas, 2024-07-09:
> @strongiukas This integration only publishes the modbus registers that Huawei solar devices make available. From what your saying your country does not provide 'Nett Metering', but charges per [...]

You said, what I didn't want to hear :( 
But anyway.
On your question: they are charging me for 10kWh and then separately paying me for 5kWh. 

But in our country we still have very good conditions. we can choose between 3 main plans:
1. I give 32%of my produced energy to grid, and 68% I can use for free.
2. I pay 4Eur for Power (if I have 10kW solar system, then 40 EUR per month) and I can use my produces energy for free.
3. I give all electricity to grid, but if I want to take it back, I have to pay 0.07Eur.
Like I said, we have really good conditions, but I don't think it will last for long. My guess would be 2-4 years.

I think we have and 4th opportunity to sell everything, but I didn't pay attention to that plan.

Roving-Ronin, 2024-07-09:
Which country is this?

If there a link you can provide to the different plans available?

wlcrs, 2024-07-21:
All relevant power meter registers have already been made available.

closed 2024-07-21 by wlcrs