wlcrs/huawei_solar #769: [Bug]:  "Add support for user-defined extra sensors" seems to be way out of date

BradleyFord, 2024-07-15:
### Describe the issue

I am struggling to replicate due to the large number of underlying code changes, if is possible to update this commit to be a better reflection of the current changes require?

https://github.com/wlcrs/huawei_solar/commit/f2b98c90fac767d2d837e81b5b8f4ecfaf480464

### Describe your Huawei Solar Setup

Inverter Type: 
Inverter Firmware version: 
sDongle present: Yes / No
sDongle Type: sDongleA-05 (WiFi / Ethernet) / SDongleA-03 (4G)
sDongle Connectivitiy: WiFi / Ethernet / 4G
sDongle Firmware: 
Power meter present: three phase / single phase / no
Optimizers Present: Yes / No
Battery: LUNA2000-SO xxkWh / LUNA2000-S1 xxkWh (Released 2024) / LG RESU xxkWh / None
Battery Firmware version: 
Huawei Solar integration version: 

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
none
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

Roving-Ronin, 2024-07-17:
@BradleyFord if its of assistance, have a look at this current pull request: https://github.com/wlcrs/huawei_solar/pull/768
This was adding 5 sensors (and removing 2 that give incorrect readings). With the Library ( https://github.com/wlcrs/huawei-solar-lib ) already defining these modbus registers, it is a matter of adding them into the sensor.py, strings.json and language files. You do need to keep the sensors in the same order as the corresponding register for them is listed in the library.

wlcrs, 2024-07-21:
I've removed the outdated branch and did a small update to the wiki. But mainly, as Roving-Ronin said: look at #768 to see how you can add extra sensors.

closed 2024-07-21 by wlcrs