wlcrs/huawei_solar #770: [Feature Request]: service to restart the inverter remotely

Joao-Sousa-71, 2024-07-16:
### Describe your feature request

From time to time (some weeks) the inverter needs to be restarted and this is possible to do with the installer account as described here:

https://forum.huawei.com/enterprise/en/huawei-pv-grid-system/thread/667282678912794625-667213868771979264

Would  be possible to expose a service to HA ( when the elevate permissions are enabled) to perform the restart? This is extremely useful if we are abroad and will allow to create some automations.
Thank you!

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

Roving-Ronin, 2024-07-17:
@Joao-Sousa-71 why not just setup a Home Assistant automation, with whatever criteria (every 2 weeks, or if X = Y then..) as the trigger, that then after a pause turn back on ? 

For example this automation say to trigger at 3am, on every second Sunday, with the action of turning off the inverter, then waiting 5 minutes and then turning it back on.

```
description: Huawei - Turn inverter off, wait 5min, then turn on
mode: single
trigger:
  - platform: time
    at: "03:00:00"
condition:
  - condition: template
    value_template: |
      {{ (now().isoweekday() == 7) and ((now().day - 1) // 7) % 2 == 1 }}
action:
  - service: switch.turn_off
    target:
      entity_id: switch.inverter_inverter_on_off
    data: {}
  - delay:
      minutes: 5
  - service: switch.turn_on
    target:
      entity_id: switch.inverter_inverter_on_off
    data: {}

```

Joao-Sousa-71, 2024-07-18:
@Roving-Ronin that is a possible workaround. I didn’t thought that turning off the Inverter will still keep the device on the network but that is something that I will try. Anyway if it not to much effort the feature requested is something that can be useful for remote management.

wlcrs, 2024-07-21:
The Inverter ON/OFF switch which is already available in this integration is the only way to start/stop the inverter that is documented in the Modbus interface.

Thank you @Roving-Ronin for explaining how this can be used in an automation.

Note that I personally think that this feature should be used with care, certainly when the inverter is under load. It causes mechanical wear to the relays in the inverter, and can lead to early failure if frequently used.

closed 2024-07-21 by wlcrs