wlcrs/huawei_solar #778: [Bug]: Integration fails to start, 'HuaweiSolarBridge' object has no attribute 'software_version'.

Toomoch, 2024-07-28:
### Describe the issue

Integration fails to start with serial RS485.

```
Error setting up entry SUN2000-4KTL-L1 for huawei_solar

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 586, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/__init__.py", line 238, in async_setup_entry
    raise err
  File "/config/custom_components/huawei_solar/__init__.py", line 104, in async_setup_entry
    primary_bridge_device_infos = await compute_device_infos(
                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/__init__.py", line 284, in compute_device_infos
    sw_version=bridge.software_version,
               ^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'HuaweiSolarBridge' object has no attribute 'software_version'. Did you mean: 'firmware_version'?
```

### Describe your Huawei Solar Setup

Inverter Type: SUN2000-4KTL-L1 
Inverter Firmware version: V200R001C00SPC130 
sDongle present: No
sDongle Type: N/A
sDongle Connectivitiy: N/A
sDongle Firmware: N/A
Power meter present: single phase
Optimizers Present: No
Battery: None
Battery Firmware version: N/A
Huawei Solar integration version: 1.4.1

### How do you connect to the inverter?

Over serial, with a serial-to-USB stick

### Upload your Diagnostics File

Cannot download diagnostics as I can't set up the device, it worked previously but as a troubleshooting step I deleted the integration.
[image]
The option is not present.

### Upload your relevant debug logs

```text
2024-07-28 12:41:13.269 DEBUG (MainThread) [homeassistant.bootstrap] Running timeout Zones: {'huawei_solar': <huawei_solar: 0 / 1>}
2024-07-28 12:41:14.270 DEBUG (MainThread) [homeassistant.bootstrap] Waiting on tasks: {<Task pending name='setup component huawei_solar' coro=<async_setup_component() running at /usr/src/homeassistant/homeassistant/setup.py:167> wait_for=<Future pending cb=[_chain_future.<locals>._call_check_cancel() at /usr/local/lib/python3.12/asyncio/futures.py:387, Task.task_wakeup()]> cb=[set.remove(), gather.<locals>._done_callback() at /usr/local/lib/python3.12/asyncio/tasks.py:767]>}
2024-07-28 12:41:14.270 DEBUG (MainThread) [homeassistant.bootstrap] Running timeout Zones: {}
2024-07-28 12:41:14.303 DEBUG (MainThread) [homeassistant.loader] Component huawei_solar import took 0.058 seconds (loaded_executor=True)
2024-07-28 12:41:14.303 INFO (MainThread) [homeassistant.setup] Setting up huawei_solar
2024-07-28 12:41:14.303 INFO (MainThread) [homeassistant.setup] Setup of domain huawei_solar took 0.00 seconds
2024-07-28 12:41:14.303 DEBUG (MainThread) [pymodbus.logging] Connecting to /dev/ttyUSB0.
2024-07-28 12:41:14.303 DEBUG (MainThread) [pymodbus.logging] Connecting comm
2024-07-28 12:41:14.305 DEBUG (MainThread) [pymodbus.logging] Connected to comm
2024-07-28 12:41:14.305 DEBUG (MainThread) [pymodbus.logging] callback_connected called
2024-07-28 12:41:14.305 DEBUG (MainThread) [huawei_solar.modbus] Waiting for 1500 milliseconds after connection before performing operations

------------CUT-------------------

2024-07-28 12:41:15.271 DEBUG (MainThread) [homeassistant.bootstrap] Running timeout Zones: {}
2024-07-28 12:41:15.306 INFO (MainThread) [huawei_solar.huawei_solar] Waiting for connection ...
2024-07-28 12:41:15.806 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 43006 with length 1 from slave 1
2024-07-28 12:41:15.806 DEBUG (MainThread) [pymodbus.logging] send: 0x1 0x3 0xa7 0xfe 0x0 0x1 0xc6 0x8e
2024-07-28 12:41:15.807 DEBUG (MainThread) [pymodbus.logging] Adding transaction 1
2024-07-28 12:41:15.838 DEBUG (MainThread) [pymodbus.logging] recv: 0x1 0x3 0x2 0x0 0x3c 0xb8 0x55 old_data:  addr=None
2024-07-28 12:41:15.838 DEBUG (MainThread) [pymodbus.logging] Processing: 0x1 0x3 0x2 0x0 0x3c 0xb8 0x55
[... 106 lines cut ...]
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 586, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/__init__.py", line 238, in async_setup_entry
    raise err
  File "/config/custom_components/huawei_solar/__init__.py", line 104, in async_setup_entry
    primary_bridge_device_infos = await compute_device_infos(
                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/__init__.py", line 284, in compute_device_infos
    sw_version=bridge.software_version,
               ^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'HuaweiSolarBridge' object has no attribute 'software_version'. Did you mean: 'firmware_version'?
2024-07-28 12:41:16.977 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event component_loaded[L]: component=huawei_solar>
2024-07-28 12:41:16.978 DEBUG (MainThread) [homeassistant.bootstrap] Waiting for startup to wrap up
2024-07-28 12:41:16.979 DEBUG (MainThread) [homeassistant.bootstrap] Integration setup times: {'huawei_solar': 2.6642438630078686, 'hacs': 1.4370778180018533, 'hardware': 1.1356068379973294, 'som_energia': 1.0029841100040358, 'melcloud': 0.7204731960082427, 'radio_browser': 0.2613149719982175, 'met': 0.22083844598091673, 'recorder': 0.0572947079926962, 'logger': 0.04250602799584158, 'system_log': 0.0420256409997819, 'energy': 0.02725787099916488, 'homeassistant': 0.026912186018307693, 'bluetooth': 0.024203016990213655, 'http': 0.008089933995506726, <Platform.TTS: 'tts'>: 0.006593190992134623, 'zeroconf': 0.0053353039984358475, 'ssdp': 0.004769674007548019, 'frontend': 0.0045077200047671795, 'mobile_app': 0.004471348016522825, 'stream': 0.0043997690081596375, 'sun': 0.003459448998910375, 'network': 0.00335516499762889, 'cloud': 0.003353010004502721, 'timer': 0.003145719994790852, 'config': 0.00242214000900276, 'shopping_list': 0.002255031999084167, 'input_select': 0.0022214240016182885, 'zone': 0.0022118820052128285, 'person': 0.0020450839947443455, 'counter': 0.0016928520053625107, <Platform.CLIMATE: 'climate'>: 0.0016436260048067197, 'input_boolean': 0.0011596200056374073, <Platform.TODO: 'todo'>: 0.0011227259965380654, 'lovelace': 0.0010269869962939993, 'api': 0.0010144050029339269, 'input_number': 0.0009775710059329867, 'scene': 0.0009427899931324646, 'auth': 0.0009376419911859557, 'persistent_notification': 0.0008625350019428879, 'schedule': 0.0008187799976440147, 'image_upload': 0.000794538005720824, 'input_button': 0.0007871679990785196, 'input_text': 0.0007694889936828986, 'assist_pipeline': 0.0007400159956887364, 'media_source': 0.0007324790058191866, 'automation': 0.0007024829974398017, 'diagnostics': 0.0006076439894968644, 'input_datetime': 0.0006065860070521012, 'logbook': 0.0005877749936189502, 'script': 0.0005301519995555282, 'history': 0.0005168170027900487, 'conversation': 0.0005081190029159188, 'tag': 0.0004645950102712959, <Platform.WATER_HEATER: 'water_heater'>: 0.00038848900294397026, 'backup': 0.00038799099274910986, 'google_translate': 0.0003855299873976037, <Platform.NOTIFY: 'notify'>: 0.00036288099363446236, 'dhcp': 0.00035996100632473826, 'stt': 0.0003506679931888357, 'application_credentials': 0.00034041500475723296, 'intent': 0.00032767700031399727, 'webhook': 0.00031376699917018414, 'repairs': 0.00030632500420324504, <Platform.DEVICE_TRACKER: 'device_tracker'>: 0.00030517400591634214, <Platform.WEATHER: 'weather'>: 0.00028364099853206426, 'analytics': 0.00023970699112396687, 'ffmpeg': 0.00016910200065467507, 'system_health': 0.00015274900943040848, 'homeassistant_alerts': 0.0001450469862902537, 'websocket_api': 0.00014145899331197143, 'usb': 0.0001343809999525547, 'sensor': 0.00012576099834404886, 'trace': 0.00011960198753513396, 'my': 8.810700091999024e-05, 'file_upload': 8.802599040791392e-05, <Platform.BINARY_SENSOR: 'binary_sensor'>: 8.731099660508335e-05, 'map': 8.445799176115543e-05, 'onboarding': 8.217299182433635e-05, 'wake_word': 5.9115991462022066e-05, 'blueprint': 4.121000529266894e-05, 'device_automation': 3.322599513921887e-05, 'search': 2.7237998438067734e-05, 'default_config': 2.524800947867334e-05}
2024-07-28 12:41:16.979 INFO (MainThread) [homeassistant.bootstrap] Home Assistant initialized in 7.70s
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

booze87, 2024-08-07:
Same here. Happened since I installed Core 2024.7.4

Toomoch, 2024-08-15:
Same issue in Core 2024.8.1

carlosdominique, 2024-09-01:
Is the issue solved or it’s occurring also in Core 2024.8.3?

Toomoch, 2024-09-29:
> Is the issue solved or it’s occurring also in Core 2024.8.3?

Same issue still.

Toomoch, 2024-10-30:
This is still an issuw.

wlcrs, 2024-10-30:
This error indicates that an incorrect version of the underlying `huawei-solar` Python library is installed. 

I suspect that this is caused by your HA installation containing another integration which wants to use an incompatible version of pyModbus, which is also dependency of the `huawei-solar` Python library. The Python package installer tries to resolve this conflict by downgrading the version of `huawei-solar` that it installs.

**TL;DR**: you run into a compatibility issue with another integration which uses the same communication library, causing incorrect software versions to be installed. 

The only way to solve this is remove the conflicting integration(s) and re-install this integration to force the installation of the correct pyModbus and `huwei-solar` library.

As this is not a bug in this integration, but a limitation of how Python library dependencies are installed, I am closing this issue. I will only revisit this issue if you can prove it still occurs in a fresh HA installation with only this integration installed.