wlcrs/huawei_solar #790: [Bug]: Integration fails with HA 2024.8

Tuningmormota, 2024-08-08:
### Describe the issue

After update to 2024.8 the integration can not connect.

### Describe your Huawei Solar Setup

Inverter Type: sun2000-2KTL-Lo
Inverter Firmware version: 130
sDongle present: No
sDongle Type: 
sDongle Connectivitiy: 
sDongle Firmware: 
Power meter present:  no
Optimizers Present: Yes
Battery:None
Battery Firmware version: 
Huawei Solar integration version:  1.4.1

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

Connected to inverter with Local Operation and Maintainance activarted, and through port 6607 while the inverter is connected to the router Wifi Network. (option was not selectable from the list above).

### Upload your relevant debug logs

```text
2024-08-08 10:57:03.197 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error.
Traceback (most recent call last):
File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
return await fut
^^^^^^^^^
File "/usr/local/lib/python3.12/asyncio/locks.py", line 212, in wait
await fut
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 199, in create
await huawei_solar._initialize()
File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 118, in _initialize
self.time_zone = (await self.get(rn.TIME_ZONE)).value
^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 271, in get
return (await self.get_multiple([name], slave))[0]
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 308, in get_multiple
response = await self._read_registers(
^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 444, in _read_registers
async with self._communication_lock():
File "/usr/local/lib/python3.12/contextlib.py", line 210, in __aenter__
return await anext(self.gen)
^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 157, in _communication_lock
await asyncio.wait_for(
File "/usr/local/lib/python3.12/asyncio/tasks.py", line 519, in wait_for
async with timeouts.timeout(timeout):
File "/usr/local/lib/python3.12/asyncio/timeouts.py", line 115, in __aexit__
raise TimeoutError from exc_val
TimeoutError
2024-08-08 10:57:06.258 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 113] Connect call failed ('192.168.2.16', 6607)
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

Tuningmormota, 2024-08-08:
Issue revowoked, solved with time

closed 2024-08-08 by Tuningmormota

philkunz, 2024-08-08:
How did you solve it? Having the same problem:

Logger: huawei_solar.huawei_solar
Source: /usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py:203 
First occurred: 8:02:21 PM (23 occurrences) 
Last logged: 8:26:40 PM

Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/locks.py", line 212, in wait
    await fut
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 199, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 118, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 271, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 308, in get_multiple
    response = await self._read_registers(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 444, in _read_registers
    async with self._communication_lock():
  File "/usr/local/lib/python3.12/contextlib.py", line 210, in __aenter__
    return await anext(self.gen)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 157, in _communication_lock
    await asyncio.wait_for(
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 519, in wait_for
    async with timeouts.timeout(timeout):
  File "/usr/local/lib/python3.12/asyncio/timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

Alevale, 2024-08-22:
Same issue here, what was the fix?

Tuningmormota, 2024-08-22:
In my case the issue solved itself with time. After the ha update (pymodbus update) it probably wanted to connect with a new session to the inverter. As the previous connection was still considered as active in the inverter it simply refused. After certain time the inverter closed the session which was created with the earlier version of Ha, and a new could connected. A proper restart of the inverter may speeds up this. That is why I closed the issue shortly after creation.