wlcrs/huawei_solar #791: [Bug]: Changed Battery, useless warning

TopoDiFogna, 2024-08-09:
### Describe the issue

I had to change the battery for a faulty hardware, the inverter now reports battery_1_type = 0 which is NONE as stated in huawei-solar.register_values and battery_2_type=2 whi is LUNA2000 from the same file.

The integration reports on startup that Detected two batteries of a different type. This can lead to unexpected behavior, which is true just because battery 1 became NONE

### Describe your Huawei Solar Setup

Inverter Type: SUN2000-6KTL-L1
Inverter Firmware version: V200R001C00SPC130
sDongle present: Yes / No
sDongle Type: sDongleA-05 (WiFi / Ethernet)
sDongle Connectivitiy: WiFi
sDongle Firmware: 
Power meter present:single phase 
Optimizers Present: Yes
Battery: LUNA2000-SO 6kWh
Battery Firmware version: 
Huawei Solar integration version: 1.4.1

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

[config_entry-huawei_solar-a4d413360b55e2872fb2e82e748906f3.json](https://github.com/user-attachments/files/16565070/config_entry-huawei_solar-a4d413360b55e2872fb2e82e748906f3.json)

### Upload your relevant debug logs

```text
No relevant debug logs, everything is visible from Diagnostic file
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

Roving-Ronin, 2024-08-13:
Have you also reapplied the firmware to the LUNA battery? Common issue is that when a new (or replacement) battery module is added to an existing battery stack, the new module has an older revision of firmware upon it, that they causes issues/bad readings for the overall battery.

wlcrs, 2024-10-14:
Should be fixed in v1.5.0 alpha 1 and above.

closed 2024-10-14 by wlcrs