wlcrs/huawei_solar #792: [Bug]: Documentation of supported firmware versions outdated?

mpw96, 2024-08-10:
### Describe the issue

In the readme you state that [This integration supports inverters running V**200**R001C00 firmware](https://github.com/wlcrs/huawei_solar#:~:text=This%20integration%20supports%20inverters%20running%20V200R001C00%20firmware).

However, there's issue #673. The author @hvorragend states that they are using inverter firmware version V**100**R001C00SPC159. From the documentation I take it that this is not supported. But still the integration seems to work form them and they get support. I'm asking because I also have a SUN2000-8KTL-M1 (with firmware version V**100**R001C00SPC165) and I would like to use this integration. A firmware version V**200**R001C00 does not seem to be available for it.

So is the documentation wrong/outdated and V**100**R001C00 is supported?

### Describe your Huawei Solar Setup

Inverter Type: SUN2000-8KTL-M1
Inverter Firmware version:  V100R001C00SPC165

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
not applicable
```

### Please confirm the following:

- [ ] I'm running the latest release of Home Assistant.
- [ ] I'm running the latest release of this integration.
- [ ] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

Sven-H-x, 2024-08-11:
Here are two links that may provide additional information: 
[WIKI - Confirmed firmware versions](https://github.com/wlcrs/huawei_solar/wiki/Confirmed-firmware-versions)

Or general:

To properly use the Modbus TCP function, each device subcomponent has a minimum version number. The following table lists the minimum version number requirements, as follow:

[image]

Source:
https://forum.huawei.com/enterprise/en/modbus-tcp-guide/thread/667250677153415168-667213868771979264

mpw96, 2024-08-11:
Ah, thank you @Sven-H-x, that helps me. Looks like I should give it a try. However, the readme is a little misleading for beginners as it seems to indicate that this integration works with V**200**R001C00 only.

Roving-Ronin, 2024-08-13:
Given the M1 series only has v100xxxx firmware, whilst some other model ranges have v200xxx firmware, this comment is incorrect. 

Maybe this statement:

"_This integration supports inverters running V200R001C00 firmware. If you are running an older V100R001C00 firmware, then please consult the "Upgrade guide" included with V200R001C00 firmware versions to check if your inverter is eligibile for an upgrade._"

Would be better ammended to (something like):

"_This integration supports inverters running various firmware revisions. It is recommended to review the 'Confirmed Firmware Versions' tables at https://github.com/wlcrs/huawei_solar/wiki/Confirmed-firmware-versions, finding the latest stable version of the Inverter / sDongle / MBUS / Battery firmware that has been submitted by users of this integration, and updating your own hardware to match a stable firmware version that others have found to be reliable._"

PS.  In future we'll probably need to add another row or table for users who are connecting via the EMMA device (when it becomes supported by the integration). Also worth noting addition of MBUS column, as lots of users overlook this is also a firmware that needs to be updated for compability with the other firmware updates.

wlcrs, 2024-08-13:
That statement in the README was the result of somebody having issues because he was running an outdated firmware version. I've revised the text to something more generic as it's not as black-and-white when it comes to the version prefix (V200... vs V100...).

closed 2024-08-13 by wlcrs

Roving-Ronin, 2024-08-13:
If anyone finds this of use... Just stumbled on this Huawei firmware roadmap, that details the latest firmware released and the 'production' version.

[Service Map Huawei.pdf](https://github.com/user-attachments/files/16597469/Service.Map.Huawei.pdf)