wlcrs/huawei_solar #794: [Bug]: Unit mismatch [W] vs [kWh] for register 32108 - TOTAL_DC_INPUT_POWER

heinemannj, 2024-08-12:
### Describe the issue

Unit mismatch [W] vs [kWh]:

`src/huawei_solar/register_names.py`
`TOTAL_DC_INPUT_POWER = "total_dc_input_power"`

`src/huawei_solar/registers.py`
`rn.TOTAL_DC_INPUT_POWER: U32Register("kWh", 100, 32108, 2),`

[image]

```
    HuaweiSolarSensorEntityDescription(
        key=rn.TOTAL_DC_INPUT_POWER,
        native_unit_of_measurement=UnitOfEnergy.KILO_WATT_HOUR,
        device_class=SensorDeviceClass.ENERGY,
        state_class=SensorStateClass.TOTAL,
    ),
```
Should be, but not found in actual Modbus definitions files to validate:

```
    HuaweiSolarSensorEntityDescription(
        key=rn.TOTAL_DC_INPUT_POWER,
        native_unit_of_measurement=UnitOfPower.WATT,
        device_class=SensorDeviceClass.POWER,
        state_class=SensorStateClass.MEASUREMENT,
    ),
```

Nevertheless this sensor seems to be really helpful as noted https://github.com/wlcrs/huawei_solar/discussions/355#discussioncomment-10304837

[Uploading Solar_Inverter_Modbus_Interface_Definitions_2024-07-18.pdf…]()

### Describe your Huawei Solar Setup

Inverter Type: sun2000-KTL6-M1
Inverter Firmware version: latest
sDongle present: Yes
sDongle Type: sDongleA-05 (WiFi / Ethernet)
sDongle Connectivitiy: Ethernet
sDongle Firmware: latest
Power meter present: three phase
Optimizers Present: No
Battery: LUNA2000-SO 15Wh
Battery Firmware version: latest
Huawei Solar integration version: latest

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
N/A
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2024-08-12:
I really dislike how Huawei named that register. As it is expressed in kWh, and thus depicts the **total amount** of power **over time**, it is in fact energy, and not power (which is always an instantaneous measurement).

Imho I've improved how the register is named. 

Nevertheless, I keep the erroneous naming from Huawei in the underlying library for clarity when you compare it to the PDF. I'm not going to change that key as it would break that sensor for any existing installation that has it already enabled. 

I don't understand what you mean with "unit mismatch"?

Roving-Ronin, 2024-08-13:
As Thijs has 'renamed' the sensor, is really the correct name that Huawei should have called it, given it is referring to energy (not power). 
In regards the 'unit mismatch', it think that Joerg is working on the basis that the official register being TOTAL_DC_INPUT_POWER would normally equate to it having a unit of 'W', but in this case the use of 'kWh' is correct.

Summary.. naming error by Huawei, Huawei integration is fine and nothing to change/fix.

closed 2024-08-13 by wlcrs