wlcrs/huawei_solar #800: [Bug]: POWER_VOLT_AMPERE_REACTIVE is a deprecated constant which will be removed in HA Core 2025.9

heinemannj, 2024-09-05:
### Describe the issue

Logger: homeassistant.const
Source: helpers/deprecation.py:197
First occurred: 09:35:49 (1 occurrences)
Last logged: 09:35:49

POWER_VOLT_AMPERE_REACTIVE was used from huawei_solar, this is a deprecated constant which will be removed in HA Core 2025.9. Use UnitOfReactivePower.VOLT_AMPERE_REACTIVE instead, please report it to the author of the 'huawei_solar' custom integration

[image]

### Describe your Huawei Solar Setup

Inverter Type: 
Inverter Firmware version: 
sDongle present: Yes / No
sDongle Type: sDongleA-05 (WiFi / Ethernet) / SDongleA-03 (4G)
sDongle Connectivitiy: WiFi / Ethernet / 4G
sDongle Firmware: 
Power meter present: three phase / single phase / no
Optimizers Present: Yes / No
Battery: LUNA2000-SO xxkWh / LUNA2000-S1 xxkWh (Released 2024) / LG RESU xxkWh / None
Battery Firmware version: 
Huawei Solar integration version: latest

### How do you connect to the inverter?

Please select your connection method

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
see above
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

closed 2024-09-14 by wlcrs

velodromen, 2024-10-17:
I have this issue as well, though the fix as far as I can see was released in September? What am I missing please or is is this still an issue?

martinlindjarv, 2024-11-14:
The latest release was 1.4.1 (https://github.com/wlcrs/huawei_solar/releases/tag/1.4.1) and it was 16.06.2024.
It seems that change is merged, but there has been no release after ...

PR #804 (2024-09-14): sensor: fix unit of reactive power
This fixes #800.
I believe #802 is wrong.