wlcrs/huawei_solar #801: [Feature Request]: Active Power Control Mode 'Zero Export Limitation' should be please added

max19751, 2024-09-07:
### Describe your feature request

To make sure no power is exported to the grid when dynamic prices are negative (and one would have to pay for the exported power such as in NL), the Sun2000 firmware offers the option 'Zero Export Limitation' under 'Active Power Control Mode'.
That options seems missing from the current HA implementation. Could that be added?

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

Roving-Ronin, 2024-09-08:
This sounds more like something that you need to be setting up with an automation, that has a condition of:

- If export price is below $0.0x then has an ACTION that uses the 'Huawei Solar: Limit the power fed to the grid' and set it to say (leave a small buffer) 250W maxinum.

Alternatively it looks like this service can now also be used (looks might be the Huawei control you mention) vs the prior action that is a bit of a 'manual' way.... "Huawei Solar: Set Active Power Control to 'Zero power grid connection'"

- Then another automation to say the opposite, that if your live export price goes back above $0.0x, then call the action 'Huawei Solar: Set Active Power Control to Maximum Feed Grid Power' to pump power into the grid.

All of this type of setup is up to the end user to configure, the integration gives the tools/means but the user has to work out what you want and set it up. Same as if you have LUNA batteries and an EV charger (that you can get a reading on its Active Power) its an idea to setup an automation that says 'if EV charge Active Power is upon 500W for more than 1min then change 'Batteries Maximum Discharge Power' to 0, so the EV charger doesn't just drain the LUNA batteries.

max19751, 2024-09-08:
The automation part you talk about I fully understand, my request was about adding the Active Power Control Mode “Zero Export Limitation”, next to the ones already existing in HA, such as “limit the power fed to grid” ones. 
See attached screenshot of the huawei options.
[image]

wlcrs, 2024-09-15:
[image]
Already available

closed 2024-09-15 by wlcrs