wlcrs/huawei_solar #803: [Bug]: WiFi connection, Grid connection with derating due to power rationing

codex70, 2024-09-11:
### Describe the issue

This issue is similar to https://githubissues.com/wlcrs/huawei_solar/714,  however my connection is via a WiFi card in the raspberry pi where HA runs, to the inverter. 

Since installing this plugin, I am no longer able to export any electricity to the grid. My battery is working as expected, but the inverter seems to stop generating excess electricity and sending it to the grid.

Is there any way of fixing this, either in the FusionSolar app,or within Home Assistant, as this is now a major problem. 

I haven't changed the configuration of the system from Home Assistant, my main goal was to control equipment depending on the output of the panels. 

I have 2 inverters, a dongle a battery and a backup system. I'm currently only connected to one of the inverters, but was planning on creating a bridge to the other once the system is working correctly. 

### Describe your Huawei Solar Setup

Inverter Type: 2 x SUN2000-4KTL-L1 
Inverter Firmware version: V200R001C00SPC130
sDongle present: Yes
sDongle Type: sDongleA-05 WiFi
sDongle Connectivitiy: WiFi
sDongle Firmware: V200R022C10SPC103
Power meter present: single phase
Optimizers Present: Yes / No
Battery: LUNA2000-5KW-C0
Battery Firmware version: 
Huawei Solar integration version: 

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

[config_entry-huawei_solar-01J71F66ZC4BG67V2AKQ0EMEA0.json](https://github.com/user-attachments/files/16968988/config_entry-huawei_solar-01J71F66ZC4BG67V2AKQ0EMEA0.json)

### Upload your relevant debug logs

```text
2024-09-11 21:22:50.976 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x2c 0x0 0x0 0x0 0x2b 0x0 0x3 0x28 0x0 0xa 0x0 0x2 0x0 0x5 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x2 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x29 0xa4 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 old_data:  addr=None
2024-09-11 21:22:50.977 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x2c 0x0 0x0 0x0 0x2b 0x0 0x3 0x28 0x0 0xa 0x0 0x2 0x0 0x5 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x2 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x29 0xa4 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2024-09-11 21:22:50.977 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2024-09-11 21:22:50.977 DEBUG (MainThread) [pymodbus.logging] Getting transaction 44
2024-09-11 21:22:50.977 DEBUG (MainThread) [pymodbus.logging] Very short frame (NO MBAP):  wait for more data
2024-09-11 21:22:50.978 DEBUG (MainThread) [pymodbus.logging] [10, 2, 5, 0, 0, 0, 0, 0, 0, 512, 0, 0, 0, 0, 0, 10660, 0, 0, 0, 0]
2024-09-11 21:22:50.978 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2024-09-11 21:22:50.979 DEBUG (MainThread) [huawei_solar.bridge] Batch update of the following registers: input_power, day_active_power_peak, active_power, efficiency, device_status, startup_time, shutdown_time, accumulated_yield_energy, total_dc_input_power, hourly_yield_energy, daily_yield_energy
2024-09-11 21:22:50.980 DEBUG (MainThread) [huawei_solar.huawei_solar] Sleeping for 0.047657 seconds before making next call.
2024-09-11 21:22:51.029 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064 with length 52 from slave 0
2024-09-11 21:22:51.030 DEBUG (MainThread) [pymodbus.logging] Adding transaction 45
2024-09-11 21:22:51.030 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer - 
2024-09-11 21:22:51.031 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x2d 0x0 0x0 0x0 0x6 0x0 0x3 0x7d 0x40 0x0 0x34
2024-09-11 21:22:51.053 WARNING (MainThread) [custom_components.localtuya.common] Missing device configuration for device_id bf46e2969b274c6e3acukt
2024-09-11 21:22:51.173 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x2d 0x0 0x0 0x0 0x6b 0x0 0x3 0x68 0x0 0x0 0x0 0x0 0x9 0x1b 0x0 0x0 0x0 0x0 0x4 0x96 0x0 0x2 0x0 0x0 0x0 0x0 0x5 0x42 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x11 0x30 0x0 0x0 0x1 0x2b 0xff 0xff 0xff 0xff 0x3 0xe8 0x13 0x8b 0x27 0x10 0x1 0x4d 0xb 0xb8 0x2 0x1 0x1a 0x93 0x66 0xe1 0x4c 0x98 0xff 0xff 0xff 0xff 0x0 0x0 0x1 0x2c 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0xd7 0x0 0x0 0x3b 0xc8 0x0 0x0 0x3f 0xac 0x66 0xe2 0xa 0x22 0x0 0x0 0x0 0xb 0x0 0x0 0x6 0x67 old_data:  addr=None
2024-09-11 21:22:51.173 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x2d 0x0 0x0 0x0 0x6b 0x0 0x3 0x68 0x0 0x0 0x0 0x0 0x9 0x1b 0x0 0x0 0x0 0x0 0x4 0x96 0x0 0x2 0x0 0x0 0x0 0x0 0x5 0x42 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x11 0x30 0x0 0x0 0x1 0x2b 0xff 0xff 0xff 0xff 0x3 0xe8 0x13 0x8b 0x27 0x10 0x1 0x4d 0xb 0xb8 0x2 0x1 0x1a 0x93 0x66 0xe1 0x4c 0x98 0xff 0xff 0xff 0xff 0x0 0x0 0x1 0x2c 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0xd7 0x0 0x0 0x3b 0xc8 0x0 0x0 0x3f 0xac 0x66 0xe2 0xa 0x22 0x0 0x0 0x0 0xb 0x0 0x0 0x6 0x67
2024-09-11 21:22:51.174 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2024-09-11 21:22:51.174 DEBUG (MainThread) [pymodbus.logging] Getting transaction 45
2024-09-11 21:22:51.175 DEBUG (MainThread) [pymodbus.logging] Very short frame (NO MBAP):  wait for more data
2024-09-11 21:22:51.175 DEBUG (MainThread) [pymodbus.logging] [0, 0, 2331, 0, 0, 1174, 2, 0, 0, 1346, 0, 0, 0, 0, 0, 4400, 0, 299, 65535, 65535, 1000, 5003, 10000, 333, 3000, 513, 6803, 26337, 19608, 65535, 65535, 0, 300, 0, 0, 0, 0, 0, 0, 0, 0, 471, 0, 15304, 0, 16300, 26338, 2594, 0, 11, 0, 1639]
2024-09-11 21:22:51.176 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2024-09-11 21:22:51.176 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x110']
2024-09-11 21:22:51.176 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x01+']
2024-09-11 21:22:51.177 DEBUG (MainThread) [pymodbus.logging] handle: [b'f\xe1', b'L\x98']
2024-09-11 21:22:51.177 DEBUG (MainThread) [pymodbus.logging] handle: [b'\xff\xff', b'\xff\xff']
2024-09-11 21:22:51.178 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b';\xc8']
2024-09-11 21:22:51.178 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'?\xac']
2024-09-11 21:22:51.178 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x0b']
2024-09-11 21:22:51.178 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x06g']
2024-09-11 21:22:51.179 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching HV22C0189979_inverter_data_update_coordinator data in 0.691 seconds (success: True)
2024-09-11 21:22:51.184 DEBUG (MainThread) [huawei_solar.bridge] Batch update of the following registers: power_meter_active_power, active_grid_power_factor, grid_exported_energy, grid_accumulated_energy
2024-09-11 21:22:51.185 DEBUG (MainThread) [huawei_solar.huawei_solar] Sleeping for 0.040173 seconds before making next call.
2024-09-11 21:22:51.227 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37113 with length 10 from slave 0
[... 14 lines cut ...]
2024-09-11 21:22:51.298 DEBUG (MainThread) [huawei_solar.huawei_solar] Sleeping for 0.043933 seconds before making next call.
2024-09-11 21:22:51.343 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760 with length 28 from slave 0
2024-09-11 21:22:51.343 DEBUG (MainThread) [pymodbus.logging] Adding transaction 47
2024-09-11 21:22:51.344 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer - 
2024-09-11 21:22:51.344 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x2f 0x0 0x0 0x0 0x6 0x0 0x3 0x93 0x80 0x0 0x1c
2024-09-11 21:22:51.402 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x2f 0x0 0x0 0x0 0x3b 0x0 0x3 0x38 0x3 0x70 0x0 0x0 0x0 0x2 0x11 0x8f 0xff 0xf9 0xff 0xff 0xfe 0xd5 0x0 0x0 0x13 0x88 0x0 0x0 0x13 0x88 0x0 0x0 0xc 0xb2 0x0 0x0 0xc 0xb2 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x14 0x1a 0x0 0x0 0x12 0xe6 0x0 0x0 0x1 0xb4 0x0 0x0 0x0 0x91 old_data:  addr=None
2024-09-11 21:22:51.402 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x2f 0x0 0x0 0x0 0x3b 0x0 0x3 0x38 0x3 0x70 0x0 0x0 0x0 0x2 0x11 0x8f 0xff 0xf9 0xff 0xff 0xfe 0xd5 0x0 0x0 0x13 0x88 0x0 0x0 0x13 0x88 0x0 0x0 0xc 0xb2 0x0 0x0 0xc 0xb2 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x14 0x1a 0x0 0x0 0x12 0xe6 0x0 0x0 0x1 0xb4 0x0 0x0 0x0 0x91
2024-09-11 21:22:51.403 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2024-09-11 21:22:51.403 DEBUG (MainThread) [pymodbus.logging] Getting transaction 47
[... cut ...]
2024-09-11 21:22:51.405 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x91']
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

Roving-Ronin, 2024-09-11:
As per the issue you referenced, this is nothing to do with the Huawei intergration or the 'Grid-Connected, Grid connection with derating due to power rationing' status. That message is due to issues with your Utility Network Provider (voltages etc) and/or the 'Grid Code' that you have set. There should be a grid code specific to your country and/or Utility Provider network, that MUST be used (don't try and fiddle the settings in the inverter or use the incorrect codes, this is illegal and could break your setup).

First thing though, connect your second inverter to the first (that I assume has the DDSU-666H meter and LUNA battery attached to it?) that will be your primary inverter that controls the secondary / overall site. This is a simple 2 wire connection between the RS485A / B ports on the inverters, a run of Cat5A cable wire is fine to use, if you don't have any official RS485 wire.

One simple thing to do is go to Settings --> Actions --> and select "Huawei Solar: Set Active Power Control to Maximum Feed Grid Power" then choose the primary inverter and select 'Perform Action'.  That shouold tell the inverter to export the max to the grid, your battery should have its settings that also determine what energy goes to it first.

If your utility provider has a limit on how may kW you can export, then use 'Huawei Solar: Limit the power fed to the grid' and enter the Watts instead.

codex70, 2024-09-12:
Thank you for your help. I didn't install the system, so don't have an installer account on FusionSolar. The second inverter is connected to the first one  and everything appeared to be working correctly when the installers left. 

I setup the plugin for HomeAssistant a couple of days after they left, but only had 1 WiFi card so could only connect to one of the inverters from HA. I'm assuming that I need to connect to the other inverter as well to get all the electricity generation figures as what I'm seeing in HA doesn't match FusionSolar. It was at this point I noticed that energy wasn't going back to the grid, so is probably coincidental.

I will check to see what settings I can change in FusionSolar and respond with more details.

Roving-Ronin, 2024-09-12:
If your connecting via the sDongle, plugged into the primary inverter, then the RS485 cable to the other inverter will provide the communications for ALL the inverters/battery/meter.

You need the installer account to get this running correctly. It's your property ask/tell the person who installed that you require it.

codex70, 2024-09-12:
When I tried to connect to the system from HA, I only saw the WiFi for the 2 inverters, so connected to the primary inverter. The system has a dongle, but I didn't see the WiFi ssid for that. Should the dongle have its own WiFi, and should I connect directly to that, or should I be getting all the info from the primary inverter? Sorry if It's a stupid question but the whole Solar stuff is new to me.

I'm currently away for the weekend, and trying to set this up remotely, I may have to wait until I get back next week to set it up properly.

Roving-Ronin, 2024-09-12:
Your connecting to the inbuilt WiFi of the inverter, if you have the sDongle you don't connect to it. You use the SolarFusion app (the one on a mobile device, not the website portal) to configure the inverter/devices, including setting up the dongle and telling IT to connect to your home WiFi network. 
If you take the bottom off the sDongle, you'll also find an RJ45 port that you can use to instead connect it to your home LAN via ethernet. Either alternative is configured via the FusionSolar app (with installer access).

codex70, 2024-09-13:
Thanks so much for your help. I'm back home early next week and will see what I can sort out.  I did ask the installer to open O&M access to the system, but they didn't know what I meant.

If they haven't changed the password, do you know if I can connect to the system with my own installer account without blocking the company from their access? 

It looks like this definitely isn't a bug, would you like me to close this ticket and comment afterwards with any information that may be useful for other users? I