wlcrs/huawei_solar #805: [Bug]: values goes to 0 and back and Go back to 0

elilei123, 2024-09-14:
### Describe the issue

Hello,
My inverter gives me the Value 5000w Then 0 Then 4900W …… what is wrong 
This is After Update 
The Same by Meter data 

### Describe your Huawei Solar Setup

Inverter Type: Sun 8 KTL
Inverter Firmware version: NEWEST 
sDongle present: Yes 
sDongle Type: sDongleA-05 (WiFi 
sDongle Connectivitiy: WiFi 
sDongle Firmware:  NEWEST WITH LIVE DATA 
Power meter present: three phase 
Optimizers Present: No
Battery: LUNA2000-SO 10kWh 
Battery Firmware version: newest
Huawei Solar integration version: 

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[image]
[image]

### Upload your relevant debug logs

```text
Nothing is Loading
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [ ] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2024-09-15:
Lowering the time between updates is at your own risk. I don't provide support for that.

closed 2024-09-15 by wlcrs

elilei123, 2024-09-15:
ah okay but what can i do