wlcrs/huawei_solar #808: [Bug]: requested discharge power isn't reflected by the battery

RikBast, 2024-09-15:
### Describe the issue

I am using EMHASS for energy management of my Huawei 3-phase inverter and battery stack. It calculates charge and discharge values which are beings send to the inverter/battery, however the values for example discharge power isn't reflected in the battery (see screenshots). The battery's discharge power doesn't comply with the request.

[image]
[image]
[image]

It seems a bug?

### Describe your Huawei Solar Setup

Inverter Type: SUN2000-6KTL-M1
Inverter Firmware version: V100R001C00SPC162
sDongle present: Yes 
sDongle Type: sDongleA-05 (WiFi / Ethernet)
sDongle Connectivitiy: Ethernet
sDongle Firmware: V100R001C00SPC133
Power meter present: three phase
Optimizers Present: Yes
Battery: LUNA2000-SO 15kWh
Battery Firmware version: V100R002C00SPC127 
Huawei Solar integration version: 1.4.1

### How do you connect to the inverter?

Please select your connection method

### Upload your Diagnostics File

[config_entry-huawei_solar-3c3424ab485c108a39e65e6c0195581a.json](https://github.com/user-attachments/files/17004893/config_entry-huawei_solar-3c3424ab485c108a39e65e6c0195581a.json)

### Upload your relevant debug logs

```text
2024-09-15 09:07:35.179 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching TA2240047133_inverter_data_update_coordinator data in 2.171 seconds (success: True)
2024-09-15 09:07:36.939 DEBUG (MainThread) [huawei_solar.bridge] Batch update of the following registers: storage_state_of_capacity, storage_running_status, storage_bus_voltage, storage_bus_current, storage_charge_discharge_power, storage_total_charge, storage_total_discharge, storage_current_day_charge_capacity, storage_current_day_discharge_capacity
2024-09-15 09:07:36.942 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760 with length 28 from slave 1
2024-09-15 09:07:36.943 DEBUG (MainThread) [pymodbus.logging] Adding transaction 454
2024-09-15 09:07:36.943 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer - 
2024-09-15 09:07:36.943 DEBUG (MainThread) [pymodbus.logging] send: 0x1 0xc6 0x0 0x0 0x0 0x6 0x1 0x3 0x93 0x80 0x0 0x1c
2024-09-15 09:07:37.185 DEBUG (MainThread) [pymodbus.logging] recv: 0x1 0xc6 0x0 0x0 0x0 0x3b 0x1 0x3 0x38 0x0 0x82 0x0 0x0 0x0 0x2 0x17 0x63 0x0 0x1 0xff 0xff 0xff 0xf7 0x0 0x0 0x3a 0x98 0x0 0x0 0x3a 0x98 0x0 0x0 0x3a 0x98 0x0 0x0 0x3a 0x98 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x7 0xc5 0xa6 0x0 0x7 0xb0 0x3 0x0 0x0 0x0 0xb 0x0 0x0 0x0 0xac old_data:  addr=None
2024-09-15 09:07:37.186 DEBUG (MainThread) [pymodbus.logging] Processing: 0x1 0xc6 0x0 0x0 0x0 0x3b 0x1 0x3 0x38 0x0 0x82 0x0 0x0 0x0 0x2 0x17 0x63 0x0 0x1 0xff 0xff 0xff 0xf7 0x0 0x0 0x3a 0x98 0x0 0x0 0x3a 0x98 0x0 0x0 0x3a 0x98 0x0 0x0 0x3a 0x98 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x7 0xc5 0xa6 0x0 0x7 0xb0 0x3 0x0 0x0 0x0 0xb 0x0 0x0 0x0 0xac
2024-09-15 09:07:37.186 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2024-09-15 09:07:37.186 DEBUG (MainThread) [pymodbus.logging] Getting transaction 454
2024-09-15 09:07:37.186 DEBUG (MainThread) [pymodbus.logging] Very short frame (NO MBAP):  wait for more data
2024-09-15 09:07:37.187 DEBUG (MainThread) [pymodbus.logging] [130, 0, 2, 5987, 1, 65535, 65527, 0, 15000, 0, 15000, 0, 15000, 0, 15000, 0, 0, 0, 0, 0, 7, 50598, 7, 45059, 0, 11, 0, 172]
2024-09-15 09:07:37.187 DEBUG (MainThread) [pymodbus.logging] handle: [b'\xff\xff', b'\xff\xf7']
2024-09-15 09:07:37.187 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x07', b'\xc5\xa6']
2024-09-15 09:07:37.187 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x07', b'\xb0\x03']
2024-09-15 09:07:37.187 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x0b']
2024-09-15 09:07:37.187 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\xac']
2024-09-15 09:07:37.188 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching TA2240047133_battery_data_update_coordinator data in 0.250 seconds (success: True)
2024-09-15 09:07:59.676 DEBUG (MainThread) [huawei_solar.bridge] Batch update of the following registers: power_meter_active_power, active_grid_power_factor, active_grid_frequency, grid_exported_energy, grid_accumulated_energy, active_grid_A_power, active_grid_B_power, active_grid_C_power
2024-09-15 09:07:59.678 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37113 with length 25 from slave 1
2024-09-15 09:07:59.679 DEBUG (MainThread) [pymodbus.logging] Adding transaction 455
2024-09-15 09:07:59.679 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer - 
2024-09-15 09:07:59.679 DEBUG (MainThread) [pymodbus.logging] send: 0x1 0xc7 0x0 0x0 0x0 0x6 0x1 0x3 0x90 0xf9 0x0 0x19
2024-09-15 09:08:00.734 DEBUG (MainThread) [pymodbus.logging] recv: 0x1 0xc7 0x0 0x0 0x0 0x35 0x1 0x3 0x32 0x0 0x0 0x2 0xd6 0x0 0x0 0x0 0xac 0xfc 0xc8 0x13 0x88 0x0 0x8 0xf9 0xe2 0x0 0x4 0xa0 0x68 0x0 0x0 0x0 0x0 0x0 0x1 0x0 0x0 0xf 0x7f 0x0 0x0 0xf 0x84 0x0 0x0 0xf 0x6c 0x0 0x0 0x0 0xd8 0x0 0x0 0x1 0x3c 0x0 0x0 0x0 0xc1 old_data:  addr=None
2024-09-15 09:08:00.734 DEBUG (MainThread) [pymodbus.logging] Processing: 0x1 0xc7 0x0 0x0 0x0 0x35 0x1 0x3 0x32 0x0 0x0 0x2 0xd6 0x0 0x0 0x0 0xac 0xfc 0xc8 0x13 0x88 0x0 0x8 0xf9 0xe2 0x0 0x4 0xa0 0x68 0x0 0x0 0x0 0x0 0x0 0x1 0x0 0x0 0xf 0x7f 0x0 0x0 0xf 0x84 0x0 0x0 0xf 0x6c 0x0 0x0 0x0 0xd8 0x0 0x0 0x1 0x3c 0x0 0x0 0x0 0xc1
2024-09-15 09:08:00.734 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2024-09-15 09:08:00.735 DEBUG (MainThread) [pymodbus.logging] Getting transaction 455
2024-09-15 09:08:00.735 DEBUG (MainThread) [pymodbus.logging] Very short frame (NO MBAP):  wait for more data
2024-09-15 09:08:00.735 DEBUG (MainThread) [pymodbus.logging] [0, 726, 0, 172, 64712, 5000, 8, 63970, 4, 41064, 0, 0, 1, 0, 3967, 0, 3972, 0, 3948, 0, 216, 0, 316, 0, 193]
2024-09-15 09:08:00.736 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x02\xd6']
2024-09-15 09:08:00.736 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x08', b'\xf9\xe2']
2024-09-15 09:08:00.736 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x04', b'\xa0h']
2024-09-15 09:08:00.736 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\xd8']
2024-09-15 09:08:00.736 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x01<']
2024-09-15 09:08:00.736 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\xc1']
[... 131 lines cut ...]
2024-09-15 09:08:39.316 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x11', b'nG']
2024-09-15 09:08:39.316 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x0e']
2024-09-15 09:08:39.316 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\xb6']
2024-09-15 09:08:39.317 DEBUG (MainThread) [huawei_solar.bridge] Batch update of the following registers: nb_online_optimizers
2024-09-15 09:08:39.317 DEBUG (MainThread) [huawei_solar.huawei_solar] Sleeping for 0.047652 seconds before making next call.
2024-09-15 09:08:39.366 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37201 with length 1 from slave 1
2024-09-15 09:08:39.367 DEBUG (MainThread) [pymodbus.logging] Adding transaction 464
2024-09-15 09:08:39.367 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer - 
2024-09-15 09:08:39.367 DEBUG (MainThread) [pymodbus.logging] send: 0x1 0xd0 0x0 0x0 0x0 0x6 0x1 0x3 0x91 0x51 0x0 0x1
2024-09-15 09:08:40.327 DEBUG (MainThread) [pymodbus.logging] recv: 0x1 0xd0 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x18 old_data:  addr=None
2024-09-15 09:08:40.327 DEBUG (MainThread) [pymodbus.logging] Processing: 0x1 0xd0 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x18
2024-09-15 09:08:40.327 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2024-09-15 09:08:40.327 DEBUG (MainThread) [pymodbus.logging] Getting transaction 464
2024-09-15 09:08:40.327 DEBUG (MainThread) [pymodbus.logging] Very short frame (NO MBAP):  wait for more data
2024-09-15 09:08:40.328 DEBUG (MainThread) [pymodbus.logging] [24]
2024-09-15 09:08:40.328 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching TA2240047133_inverter_data_update_coordinator data in 3.325 seconds (success: True)
```

### Please confirm the following:

- [ ] I'm running the latest release of Home Assistant.
- [ ] I'm running the latest release of this integration.
- [ ] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2024-09-15:
The first sensor shows that the requested (dis)charge value was correctly sent to the inverter, as the integration can read back the values that it sent to it.

How the battery handles that request is something else, but I cannot help you with that. I suggest that you reach out to Huawei.

closed 2024-09-15 by wlcrs

RikBast, 2024-09-16:
Ok, thank you. I will contact Huawei Support