wlcrs/huawei_solar #809: [Bug]: Errore improvviso integrazione

SimoneRullo, 2024-09-21:
### Describe the issue

Di seguito il dettaglio recuperato dai registri di Home Assistant.

Questo errore ha avuto origine da un'integrazione personalizzata.

Registratore: pymodbus.logging
Fonte: custom_components/huawei_solar/__init__.py:89
Integrazione: Huawei Solar (documentazione, problemi)
Prima occorrenza: 07:53:37 (16 occorrenze)
Ultima registrazione: 08:08:27

Failed to connect

Registratore: huawei_solar.huawei_solar
Fonte: /usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py:162
Prima occorrenza: 07:53:42 (16 occorrenze)
Ultima registrazione: 08:08:32

Timeout while waiting for connection. Reconnecting...
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/locks.py", line 212, in wait
    await fut
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 157, in _communication_lock
    await asyncio.wait_for(
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 519, in wait_for
    async with timeouts.timeout(timeout):
  File "/usr/local/lib/python3.12/asyncio/timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

Registratore: huawei_solar.huawei_solar
Fonte: /usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py:203
Prima occorrenza: 07:53:42 (16 occorrenze)
Ultima registrazione: 08:08:32

Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/locks.py", line 212, in wait
    await fut
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 199, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 118, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 271, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 308, in get_multiple
    response = await self._read_registers(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 444, in _read_registers
    async with self._communication_lock():
  File "/usr/local/lib/python3.12/contextlib.py", line 210, in __aenter__
[... 202 lines cut ...]
  File "/usr/src/homeassistant/homeassistant/runner.py", line 189, in run
    return loop.run_until_complete(setup_and_run_hass(runtime_config))
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 674, in run_until_complete
    self.run_forever()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 641, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1990, in _run_once
    handle._run()
  File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2184, in async_forward_entry_setup
    _report_non_awaited_platform_forwards(entry, "async_forward_entry_setup")
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 1201, in _report_non_awaited_platform_forwards
    report(
  File "/usr/src/homeassistant/homeassistant/helpers/frame.py", line 151, in report
    _
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

Javiteyu, 2024-09-22:
I have the same problem since some days ago. Raspberry PI5.

wlcrs, 2024-09-22:
This issue seems to be caused by a connection failure or an unstable connection to the inverter. This integration already has already implemented a lot of logic to automatically recover from connection failures or unstable connections. The three most important ones are:
- automatic reconnects if the connection to the inverter failed;
- each Modbus request is retried for up to 6 times, with increasing waits between each attempt to allow the connection to recover;
- after 3 failed Modbus requests the connection is reset to fix hanging connections.

Please understand that I cannot help you solve your connection failure or unstable connection. Consider [an alternative connection method](https://github.com/wlcrs/huawei_solar/wiki/Connecting-to-the-inverter).
If you are convinced that there is no connectivity issue, then I invite you to debug and fix the problem yourself and submit a pull request to the underlying [huawei-solar Python library](https://gitlab.com/Emilv2/huawei-solar). Your input would be greatly appreciated.

As there is nothing actionable for me to do in this issue, I'm closing it. You are welcome to open a topic in [Discussions: Connecting to the inverter](https://github.com/wlcrs/huawei_solar/discussions/categories/connecting-to-the-inverter) to get community support for your problem.

closed 2024-09-22 by wlcrs