wlcrs/huawei_solar #812: [Bug]: 2 Inverter but only the data from one in HA

berndan77, 2024-09-23:
### Describe the issue

there are two SUN2000-10KTL-M1 on one dongle. 16,4kwp in summary and 8,2kwp on each inverter. 
In Home Assistant i see only the data of one inverter. In Home Assistant i have always only 50% power. 
There ist only one Entität "sensor.inverter_eingangsleistung" and this Entität shows only 50%, only one inverter. 

### Describe your Huawei Solar Setup

Inverter Type:  SUN2000-10KTL-M1
Inverter Firmware version: V100R001C00SPC153
sDongle present: Yes 
sDongle Type: SDongleA-05
sDongle Connectivitiy: Ethernet 
sDongle Firmware:  V100R001C00SPC125
Power meter present: no
Optimizers Present: No
Battery: no battery
Battery Firmware version: 
Huawei Solar integration version:  1.4.1

### How do you connect to the inverter?

Please select your connection method

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[home-assistant_huawei_solar_2024-09-23T19-01-22.101Z.log](https://github.com/user-attachments/files/17102948/home
[config_entry-huawei_solar-01J7Y6TPNK4ZZ5PYYGTX3YVF5A.json](https://github.com/user-attachments/files/17102949/config_entry-huawei_solar-01J7Y6TPNK4ZZ5PYYGTX3YVF5A.json)
-assistant_huawei_solar_2024-09-23T19-01-22.101Z.log)

### Upload your relevant debug logs

```text
Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: custom_components.huawei_solar
Quelle: helpers/update_coordinator.py:360
Integration: Huawei Solar (Dokumentation, Probleme)
Erstmals aufgetreten: 21. September 2024 um 12:21:36 (2 Vorkommnisse)
Zuletzt protokolliert: 02:43:25

Timeout fetching TA2280236597_inverter_data_update_coordinator data
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

rolfch, 2024-09-29:
Hi berndan77, did you find a solution? I have the same issue...

berndan77, 2024-09-29:
[image]

Hello Rolf,

yes i find a solution. Every Inverter has an  ID. You must specify your ID`s in you Huawei Ingeration configuration. In my case the inverters have the Id`s 1 and 2.

closed 2024-10-03 by wlcrs