wlcrs/huawei_solar #813: [Feature Request]: Missing Option Field: "Charge/Discharge"

christianarth, 2024-09-26:
### Describe your feature request

On the FusionSolar website, there is an option in the battery configuration under Forced charge/discharge to change ‘Charge/Discharge’ from ‘Stop’ to ‘Charge’ or ‘Discharge’. This field is missing in Home Assistant. This toggle would be very useful to be able to charge the battery automatically depending on the electricity price in combination with ‘Charge from AC’.
Or am I overlooking this feature?

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

vdvmichel, 2024-09-29:
You can do that with service…

wlcrs, 2024-10-03:
What 👆 said

closed 2024-10-03 by wlcrs