wlcrs/huawei_solar #815: [Bug]: Daily Yield incorrect

beatschubser, 2024-09-30:
### Describe the issue

The daily yield starts building up at midnight with the inverter turned off. It seems to take the energy into account that is pulled from the battery. However, it's lower than the "daily discharge" of the battery.

Expected behavior would be to see the same daily yield as in the app, so what's actually coming from the panels, and only the panels.

### Describe your Huawei Solar Setup

Inverter Type: SUN2000-8KTL-M1
Inverter Firmware version: V100R001C00SPC159
sDongle present: Yes
sDongle Type: sDongleA-05
sDongle Connectivitiy: Ethernet
sDongle Firmware: V200R022C10SPC108
Power meter present: three phase
Optimizers Present: No
Battery: LUNA2000-SO 10kWh
Battery Firmware version: V100R002C00SPC127
Huawei Solar integration version: 1.4.1

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

[config_entry-huawei_solar-01J8X60CX2EGZCPGABW5798PR2.json](https://github.com/user-attachments/files/17193955/config_entry-huawei_solar-01J8X60CX2EGZCPGABW5798PR2.json)

### Upload your relevant debug logs

```text
2024-09-30 19:25:35.353 DEBUG (MainThread) [huawei_solar.bridge] Batch update of the following registers: power_meter_active_power, active_grid_power_factor, active_grid_frequency, grid_exported_energy, grid_accumulated_energy, active_grid_A_power, active_grid_B_power, active_grid_C_power
2024-09-30 19:25:35.354 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37113 with length 25 from slave 1
2024-09-30 19:25:35.893 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching TA2330142522_power_meter_data_update_coordinator data in 0.540 seconds (success: True)
2024-09-30 19:25:44.415 DEBUG (MainThread) [huawei_solar.bridge] Batch update of the following registers: state_1, state_2, state_3, alarm_1, alarm_2, alarm_3, pv_01_voltage, pv_01_current, pv_02_voltage, pv_02_current
2024-09-30 19:25:44.416 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000 with length 20 from slave 1
2024-09-30 19:25:45.496 DEBUG (MainThread) [huawei_solar.bridge] Batch update of the following registers: input_power, day_active_power_peak, active_power, efficiency, device_status, startup_time, shutdown_time, accumulated_yield_energy, total_dc_input_power, hourly_yield_energy, daily_yield_energy
2024-09-30 19:25:45.496 DEBUG (MainThread) [huawei_solar.huawei_solar] Sleeping for 0.049302 seconds before making next call.
2024-09-30 19:25:45.548 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064 with length 52 from slave 1
2024-09-30 19:25:46.772 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching TA2330142522_inverter_data_update_coordinator data in 2.357 seconds (success: True)
2024-09-30 19:25:46.773 DEBUG (MainThread) [huawei_solar.bridge] Batch update of the following registers: storage_state_of_capacity, storage_running_status, storage_bus_voltage, storage_bus_current, storage_charge_discharge_power, storage_total_charge, storage_total_discharge, storage_current_day_charge_capacity, storage_current_day_discharge_capacity
2024-09-30 19:25:46.773 DEBUG (MainThread) [huawei_solar.huawei_solar] Sleeping for 0.047710 seconds before making next call.
2024-09-30 19:25:46.822 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760 with length 28 from slave 1
2024-09-30 19:25:47.510 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching TA2330142522_battery_data_update_coordinator data in 1.894 seconds (success: True)
2024-09-30 19:26:05.354 DEBUG (MainThread) [huawei_solar.bridge] Batch update of the following registers: power_meter_active_power, active_grid_power_factor, active_grid_frequency, grid_exported_energy, grid_accumulated_energy, active_grid_A_power, active_grid_B_power, active_grid_C_power
2024-09-30 19:26:05.354 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37113 with length 25 from slave 1
2024-09-30 19:26:06.007 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching TA2330142522_power_meter_data_update_coordinator data in 0.653 seconds (success: True)
2024-09-30 19:26:16.416 DEBUG (MainThread) [huawei_solar.bridge] Batch update of the following registers: state_1, state_2, state_3, alarm_1, alarm_2, alarm_3, pv_01_voltage, pv_01_current, pv_02_voltage, pv_02_current
2024-09-30 19:26:16.416 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000 with length 20 from slave 1
2024-09-30 19:26:18.222 DEBUG (MainThread) [huawei_solar.bridge] Batch update of the following registers: input_power, day_active_power_peak, active_power, efficiency, device_status, startup_time, shutdown_time, accumulated_yield_energy, total_dc_input_power, hourly_yield_energy, daily_yield_energy
2024-09-30 19:26:18.222 DEBUG (MainThread) [huawei_solar.huawei_solar] Sleeping for 0.049298 seconds before making next call.
2024-09-30 19:26:18.272 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064 with length 52 from slave 1
2024-09-30 19:26:19.927 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching TA2330142522_inverter_data_update_coordinator data in 3.511 seconds (success: True)
2024-09-30 19:26:19.928 DEBUG (MainThread) [huawei_solar.bridge] Batch update of the following registers: storage_state_of_capacity, storage_running_status, storage_bus_voltage, storage_bus_current, storage_charge_discharge_power, storage_total_charge, storage_total_discharge, storage_current_day_charge_capacity, storage_current_day_discharge_capacity
2024-09-30 19:26:19.928 DEBUG (MainThread) [huawei_solar.huawei_solar] Sleeping for 0.047773 seconds before making next call.
2024-09-30 19:26:19.976 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760 with length 28 from slave 1
2024-09-30 19:26:21.193 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching TA2330142522_battery_data_update_coordinator data in 3.577 seconds (success: True)
2024-09-30 19:26:35.354 DEBUG (MainThread) [huawei_solar.bridge] Batch update of the following registers: power_meter_active_power, active_grid_power_factor, active_grid_frequency, grid_exported_energy, grid_accumulated_energy, active_grid_A_power, active_grid_B_power, active_grid_C_power
2024-09-30 19:26:35.355 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37113 with length 25 from slave 1
2024-09-30 19:26:35.597 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching TA2330142522_power_meter_data_update_coordinator data in 0.242 seconds (success: True)
2024-09-30 19:26:49.415 DEBUG (MainThread) [huawei_solar.bridge] Batch update of the following registers: state_1, state_2, state_3, alarm_1, alarm_2, alarm_3, pv_01_voltage, pv_01_current, pv_02_voltage, pv_02_current
2024-09-30 19:26:49.416 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000 with length 20 from slave 1
2024-09-30 19:26:49.684 DEBUG (MainThread) [huawei_solar.bridge] Batch update of the following registers: input_power, day_active_power_peak, active_power, efficiency, device_status, startup_time, shutdown_time, accumulated_yield_energy, total_dc_input_power, hourly_yield_energy, daily_yield_energy
2024-09-30 19:26:49.684 DEBUG (MainThread) [huawei_solar.huawei_solar] Sleeping for 0.049288 seconds before making next call.
2024-09-30 19:26:49.734 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064 with length 52 from slave 1
2024-09-30 19:26:51.122 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching TA2330142522_inverter_data_update_coordinator data in 1.707 seconds (success: True)
2024-09-30 19:26:51.617 DEBUG (MainThread) [huawei_solar.bridge] Batch update of the following registers: storage_state_of_capacity, storage_running_status, storage_bus_voltage, storage_bus_current, storage_charge_discharge_power, storage_total_charge, storage_total_discharge, storage_current_day_charge_capacity, storage_current_day_discharge_capacity
2024-09-30 19:26:51.617 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760 with length 28 from slave 1
2024-09-30 19:26:52.363 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching TA2330142522_battery_data_update_coordinator data in 0.747 seconds (success: True)
2024-09-30 19:27:05.354 DEBUG (MainThread) [huawei_solar.bridge] Batch update of the following registers: power_meter_active_power, active_grid_power_factor, active_grid_frequency, grid_exported_energy, grid_accu [... cut ...]
2024-09-30 19:30:02.415 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000 with length 20 from slave 1
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

ThomasDeMarez, 2024-10-01:
I have noticed the same thing. 
Changing between the Daily Yield and Total Yield does not seem to fix the issue, both suffer from it.

[image]

Actual solar production started between 8:00 and 9:00

wlcrs, 2024-10-03:
Duplicate of #1 (and #446, #309, #32, #155, ...)

closed 2024-10-03 by wlcrs