wlcrs/huawei_solar #818: [Bug]: Integration not loading - Error: Unable to import component: No module named 'pymodbus.message'

ragesoft, 2024-10-06:
### Describe the issue

After Update to latest home assistant core, the integration fails to load.
Restart, reinstall failed... / wont fix it. 

`Logger: homeassistant.setup
Quelle: setup.py:334
Erstmals aufgetreten: 10:23:58 (1 Vorkommnisse)
Zuletzt protokolliert: 10:23:58

Setup failed for custom integration 'huawei_solar': Unable to import component: No module named 'pymodbus.message'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1027, in async_get_component
    self._component_future.result()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1007, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1067, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/huawei_solar/__init__.py", line 20, in <module>
    from huawei_solar import (
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/__init__.py", line 3, in <module>
    from .bridge import HuaweiSolarBridge
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/bridge.py", line 28, in <module>
    from .huawei_solar import (
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 17, in <module>
    from pymodbus.message.rtu import MessageRTU
ModuleNotFoundError: No module named 'pymodbus.message'`

### Describe your Huawei Solar Setup

Inverter Type:  Sun2000-10ktl-m1 
Inverter Firmware version: 
sDongle present: Yes 
sDongle Type: sDongleA-05 (WiFi / Ethernet) 
sDongle Connectivitiy: WiFi 
sDongle Firmware: 
Power meter present: three phase
Optimizers Present:  No
Battery: LUNA2000-S0 10kWh
Battery Firmware version: 
Huawei Solar integration version: 1.4.1
Homeassistant Core: 2024.10.1
Operation System: 13.1

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

2024-10-06 10:22:56.415 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-10-06 10:23:58.230 ERROR (MainThread) [homeassistant.setup] Setup failed for custom integration 'huawei_solar': Unable to import component: No module named 'pymodbus.message'
File "/config/custom_components/huawei_solar/__init__.py", line 20, in <module>
from huawei_solar import (
File "/usr/local/lib/python3.12/site-packages/huawei_solar/__init__.py", line 3, in <module>
File "/usr/local/lib/python3.12/site-packages/huawei_solar/bridge.py", line 28, in <module>
from .huawei_solar import (
File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 17, in <module>
2024-10-06 10:25:58.630 ERROR (MainThread) [homeassistant.config_entries] Error occurred loading flow for integration huawei_solar: No module named 'pymodbus.message'
2024-10-06 10:26:06.799 ERROR (MainThread) [homeassistant.config_entries] Error occurred loading flow for integration huawei_solar: No module named 'pymodbus.message'

### Upload your relevant debug logs

```text
2024-10-06 10:22:56.415 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-10-06 10:23:58.230 ERROR (MainThread) [homeassistant.setup] Setup failed for custom integration 'huawei_solar': Unable to import component: No module named 'pymodbus.message'
File "/config/custom_components/huawei_solar/__init__.py", line 20, in <module>
from huawei_solar import (
File "/usr/local/lib/python3.12/site-packages/huawei_solar/__init__.py", line 3, in <module>
File "/usr/local/lib/python3.12/site-packages/huawei_solar/bridge.py", line 28, in <module>
from .huawei_solar import (
File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 17, in <module>
2024-10-06 10:25:58.630 ERROR (MainThread) [homeassistant.config_entries] Error occurred loading flow for integration huawei_solar: No module named 'pymodbus.message'
2024-10-06 10:26:06.799 ERROR (MainThread) [homeassistant.config_entries] Error occurred loading flow for integration huawei_solar: No module named 'pymodbus.message'
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2024-10-06:
Duplicate of #665

closed 2024-10-06 by wlcrs

bytedealer, 2024-10-06:
Hey @wlcrs,

what does „not planned“ mean? That the bug is not fixed and the integration is discontinued? Because in the state of today you cannot integrate it anymore.

Thanks in advance for a short notice
HFM

ragesoft, 2024-10-07:
@wlcrs two stupid questions:
1: how to check what modus-lib is installed on my system (and maybe "who" use it :) )

2: why is this importend dependency not marked in any manifest, if your solution requiered it to be at least pyModbus v3.6.6 (seen in other issue)? Is there any workaround so that i can manually update the lib?

Thx