wlcrs/huawei_solar #832: [Bug]: Failed to setup

Ivano62, 2024-10-21:
### Describe the issue

I've installed latest build (8e77b6c), after that I got the error in the title

### Describe your Huawei Solar Setup

Inverter Type: SUN2000-6KTL-L1
Inverter Firmware version: V200R001C00SPC144
sDongle present: Yes
sDongle Type: sDongleA-05 (WiFi / Ethernet)
sDongle Connectivitiy: Ethernet
sDongle Firmware: V200R022C10SPC121
Power meter present: single phase
Optimizers Present: Yes
Battery: LUNA2000-5KW-C0
Battery Firmware version: V100R002C00SPC627 
Huawei Solar integration version: 1.4.1 build 8e77b6c

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

2024-10-21 22:51:17.762 ERROR (ImportExecutor_0) [homeassistant.loader] Unexpected exception importing platform custom_components.huawei_solar.sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1264, in _load_platform
    cache[full_name] = self._import_platform(platform_name)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1296, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/huawei_solar/sensor.py", line 787, in <module>
    BatteryTemplateEntityDescription(
TypeError: BatteryTemplateEntityDescription.__init__() got an unexpected keyword argument 'entity_registry_enabled_default'
2024-10-21 22:51:17.820 ERROR (MainThread) [homeassistant.loader] Unexpected exception importing platform custom_components.huawei_solar.sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1264, in _load_platform
    cache[full_name] = self._import_platform(platform_name)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1296, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/huawei_solar/sensor.py", line 787, in <module>
    BatteryTemplateEntityDescription(
TypeError: BatteryTemplateEntityDescription.__init__() got an unexpected keyword argument 'entity_registry_enabled_default'
[... 128 lines cut ...]
    await hass.config_entries.async_forward_entry_setups(entry, PLATFORMS)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2147, in async_forward_entry_setups
    await integration.async_get_platforms(platforms)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1177, in async_get_platforms
    import_future.result()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1177, in async_get_platforms
    import_future.result()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1165, in async_get_platforms
    platforms.update(self._load_platforms(platform_names))
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1090, in _load_platforms
    platform_name: self._load_platform(platform_name)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1284, in _load_platform
    raise ImportError(
ImportError: Exception importing custom_components.huawei_solar.sensor
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

Illum2021, 2024-10-22:
I am not sure if this is the same problem. I can’t connect the sdongle via network as well, although  it worked stable for a year. 
I keep loosing connection and also see values for a short time every 5-20 minutes. 
tried restarting the dongle, restart the battery pack, updated dongle to latest firmware.

closed 2024-10-22 by wlcrs

Ivano62, 2024-10-22:
I don't understand why the bug was closed, the problem persists.
I restored the previous version of the integration and everything works regularly, I installed the latest one again and the problem occurs again.
The firmware of the inverter, battery and dongle are the latest available.
I activated debugging and this is the result:

2024-10-22 08:53:52.930 ERROR (ImportExecutor_0) [homeassistant.loader] Unexpected exception importing platform custom_components.huawei_solar.sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1264, in _load_platform
    cache[full_name] = self._import_platform(platform_name)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1296, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/huawei_solar/sensor.py", line 787, in <module>
    BatteryTemplateEntityDescription(
TypeError: BatteryTemplateEntityDescription.__init__() got an unexpected keyword argument 'entity_registry_enabled_default'
2024-10-22 08:53:52.990 WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to import_module with args ('custom_components.huawei_solar.sensor',) inside the event loop by custom integration 'huawei_solar' at custom_components/huawei_solar/__init__.py, line 241: await hass.config_entries.async_forward_entry_setups(entry, PLATFORMS) (offender: /usr/src/homeassistant/homeassistant/loader.py, line 1296: return importlib.import_module(f"{self.pkg_path}.{platform_name}")), please create a bug report at https://github.com/wlcrs/huawei_solar/issues
For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#import_module
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module>
    sys.exit(main())
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main
    exit_code = runner.run(runtime_conf)
  File "/usr/src/homeassistant/homeassistant/runner.py", line 189, in run
    return loop.run_until_complete(setup_and_run_hass(runtime_config))
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 674, in run_until_complete
    self.run_forever()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 641, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1990, in _run_once
    handle._run()
  File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 742, in async_setup_locked
    await self.async_setup(hass, integration=integration)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/huawei_solar/__init__.py", line 241, in async_setup_entry
    await hass.config_entries.async_forward_entry_setups(entry, PLATFORMS)

2024-10-22 08:53:53.025 ERROR (MainThread) [homeassistant.loader] Unexpected exception importing platform custom_components.huawei_solar.sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1264, in _load_platform
    cache[full_name] = self._import_platform(platform_name)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1296, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
[... 333 lines cut ...]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 628, in get_file
    return await _do_read_file()
           ^^^^^^^^^^^^^^^^^^^^^
[... cut ...] 0x0 0x80 0x79 0x0 0xe0 0x0 0x9d 0x1 0x4c 0x0 0x5 0x1 0x4 0x0 0x7b 0x2f 0xd 0x0 0xf 0x0 0x18 0x1 0x1f 0x1a 0x0 0x0 0x0 0x80 0x6e 0x0 0xff 0x0 0x98 0x1 0x4c 0x0 0x5 0x1 0x4 0x0 0xb6 0xa2 0xb 0x0
2024-10-22 08:56:39.081 DEBUG (MainThread) [pymodbus.logging] Factory Response[PrivateHuaweiModbusResponse': 65]
2024-10-22 08:56:39.082 DEBUG (MainThread) [pymodbus.logging] Getting transaction 26
2024-10-22 08:56:39.082 DEBUG (MainThread) [pymodbus.logging] Very short frame (NO MBAP):  wait for more data

Ivano62, 2024-10-22:
Hi,
sorry I copied the wrong version of the integration, now everything works regularly even if it seems to me that some labels of the BMS2 are wrong.

Thanks.

Hi,

File "", line 1387, in _gcd_import
File "", line 1360, in _find_and_load
File "", line 1331, in _find_and_load_unlocked
File "", line 935, in _load_unlocked
File "", line 995, in exec_module
File "", line 488, in _call_with_frames_removed
File "", line 198, in _run_module_as_main
File "", line 88, in _run_code
File "", line 1387, in _gcd_import
File "", line 1360, in _find_and_load
File "", line 1331, in _find_and_load_unlocked
File "", line 935, in _load_unlocked
File "", line 995, in exec_module
File "", line 488, in _call_with_frames_removed
File "", line 1387, in _gcd_import
File "", line 1360, in _find_and_load
File "", line 1331, in _find_and_load_unlocked
File "", line 935, in _load_unlocked
File "", line 995, in exec_module
File "", line 488, in _call_with_frames_removed

Thanks

Closed #832<https://github.com/wlcrs/huawei_solar/issues/832> as completed via <https://github.com/wlcrs/huawei_solar/commit/28fd020a60558c6d614b10d10e59599517ce30dd>

28fd020<https://github.com/wlcrs/huawei_solar/commit/28fd020a60558c6d614b10d10e59599517ce30dd>

-

commit 28fd020a (2024-10-22): BatteryTemplateEntityDescriptions don't need a entity_registry_enabled_default.

Fixes #832