wlcrs/huawei_solar #835: Support of SUN2000-12K-MAP0

offtopic, 2024-10-24:
### Describe your feature request

Will this also be supported in the (near) future - i know it is relatively new
I tried it out - but could not connect via network - i'm not sure if it is because of the not supported device - or maybe a fault on my end.
it is connected via IP 192.168.178.x - I tried both documented ports 6607 and 502 with slave id 0.

I'm also developer (different area - but i might help to supoort it)

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

TomHarm, 2024-10-24:
You should try it with a valid modbus id from your master inverter. 1 or 16 might do the trick. 
Is modbus enabled in the dongle?

offtopic, 2024-10-25:
@TomHarm it is working now -  i had a chat with my technician - he assured me that he had carried out the mod bus setting - but this was not the case 🤬 

So i can confirm that i can now access the values 👌:

Port: 502 
SlaveID: 1

SUN2000-12K-MAP0
Version: V200R024C00SPC100

Dongle-HV2340179967
Version: V200R022C10SPC121

closed 2024-10-25 by offtopic