wlcrs/huawei_solar #837: [Question]: some entities are disabled

0k00l, 2024-10-28:
### Describe the issue

I had every entity working - from inverter and power meter - in previous versions of HA with your integration(which is awesome btw ;) ) - i've made migration to docker with latest version of HA and HAC-s and now some entities are disabled, for example(from power meter):
- phase A voltage
- idle power
- current
- and so on

Is there a way to bring them back to live?

### Describe your Huawei Solar Setup

Inverter Type: Huawei SUN2000-6KTL-M
Inverter Firmware version: V100R001-02 (Software version V100R001C00SPC153)
sDongle present: Yes
sDongle Type: sDongleA-05 (WiFi / Ethernet)
sDongle Connectivitiy:  Ethernet
sDongle Firmware: V100R001C00SPC119
Power meter present: three phase
Optimizers Present: No
Battery: None
Battery Firmware version: 
Huawei Solar integration version: 1.4.1

### How do you connect to the inverter?

Please select your connection method

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[config_entry-huawei_solar-01JAX411RS4CWF7150767SNFR2.json](https://github.com/user-attachments/files/17543115/config_entry-huawei_solar-01JAX411RS4CWF7150767SNFR2.json)
[image]

### Upload your relevant debug logs

```text
2024-10-23 18:29:56.131 WARNING (ImportExecutor_0) [homeassistant.const] POWER_VOLT_AMPERE_REACTIVE was used from huawei_solar, this is a deprecated constant which will be removed in HA Core 2025.9. Use UnitOfReactivePower.VOLT_AMPERE_REACTIVE instead, please report it to the author of the 'huawei_solar' custom integration
2024-10-23 18:30:41.075 ERROR (MainThread) [huawei_solar.huawei_solar] Timeout while waiting for connection. Reconnecting...
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/locks.py", line 212, in wait
    await fut
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 157, in _communication_lock
    await asyncio.wait_for(
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 519, in wait_for
    async with timeouts.timeout(timeout):
  File "/usr/local/lib/python3.12/asyncio/timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError
2024-10-23 18:30:41.077 ERROR (MainThread) [custom_components.huawei_solar] Timeout fetching BT2180542143_power_meter_data_update_coordinator data
2024-10-23 18:31:16.077 ERROR (MainThread) [huawei_solar.huawei_solar] Timeout while waiting for connection. Reconnecting...
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/locks.py", line 212, in wait
    await fut
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
[... 438 lines cut ...]
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/locks.py", line 212, in wait
    await fut
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 157, in _communication_lock
    await asyncio.wait_for(
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 519, in wait_for
    async with timeouts.timeout(timeout):
  File "/usr/local/lib/python3.12/asyncio/timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError
```

### Please confirm the following:

- [ ] I'm running the latest release of Home Assistant.
- [ ] I'm running the latest release of this integration.
- [ ] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

Roving-Ronin, 2024-12-07:
@0k00l As per documentation, click on each sensor. Then click on the config icon (the set of gears / cogs up the top right), then click on the ENABLE switch for that sensor to enable.

Please close 'issue'.

wlcrs, 2025-01-21:
/This issue seems to be caused by a connection failure or an unstable connection to the inverter. This integration already has already implemented a lot of logic to automatically recover from connection failures or unstable connections. The three most important ones are:
- automatic reconnects if the connection to the inverter failed;
- each Modbus request is retried for up to 6 times, with increasing waits between each attempt to allow the connection to recover;
- after 3 failed Modbus requests the connection is reset to fix hanging connections.

Please understand that I cannot help you solve your connection failure or unstable connection. Consider [an alternative connection method](https://github.com/wlcrs/huawei_solar/wiki/Connecting-to-the-inverter).
If you are convinced that there is no connectivity issue, then I invite you to debug and fix the problem yourself and submit a pull request to the underlying [huawei-solar Python library](https://gitlab.com/Emilv2/huawei-solar). Your input would be greatly appreciated.

As there is nothing actionable for me to do in this issue, I'm closing it. You are welcome to open a topic in [Discussions: Connecting to the inverter](https://github.com/wlcrs/huawei_solar/discussions/categories/connecting-to-the-inverter) to get community support for your problem.