wlcrs/huawei_solar #838: [Question]: No configuration options for inverter/battery

m-wo, 2024-10-30:
### Describe the issue

Hi have the Huawei integration running for my 10KTL-M1 and my Luna2000-10-S0 but I don´t get the configuration options for each, as shown [here for the inverter](https://github.com/wlcrs/huawei_solar/blob/main/images/inverter_configuration_diagnostics.png) or [here for the battery](https://github.com/wlcrs/huawei_solar/blob/main/images/battery_configuration.png)

### Describe your Huawei Solar Setup

Inverter Type: Sun2000-10KTL-M1
Inverter Firmware version: V100R001C00SPC159
sDongle present: Yes
sDongle Type: sDongleA-05 (WiFi / Ethernet)
sDongle Connectivitiy: Ethernet
sDongle Firmware: V200R022C10SPC118
Power meter present: three phase
Optimizers Present: YNo
Battery: LUNA2000-SO 10kWh
Battery Firmware version: ?
Huawei Solar integration version: 1.4.1

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

[config_entry-huawei_solar-09b6630743181e48bd8695159452c66c.json](https://github.com/user-attachments/files/17579522/config_entry-huawei_solar-09b6630743181e48bd8695159452c66c.json)

### Upload your relevant debug logs

```text
Somehow I can´t upload here :/
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

m-wo, 2024-10-30:
[home-assistant_huawei_solar_2024-10-30T22-06-08.147Z.log](https://github.com/user-attachments/files/17579569/home-assistant_huawei_solar_2024-10-30T22-06-08.147Z.log)
Here is the debug log

Roving-Ronin, 2024-12-07:
@m-wo Have you done as per the documentation? https://github.com/wlcrs/huawei_solar?tab=readme-ov-file#network-configuration

"Checking the Advanced: elevate permissions checkbox..."

[image]

m-wo, 2024-12-07:
Yeah actually I missed it says that elevated permissions are reuqired, unfortunately I don't have an installer account and the company which installed my PV system said I'd loos warranty if I reset the inverter so set a new password...

closed 2024-12-07 by wlcrs