wlcrs/huawei_solar #840: [Bug]: 'Batteries Grid charge maximum power' is at max value of 'Inverter Max active power'

heinemannj, 2024-11-01:
### Describe the issue

Maximum value of `**Batteries Grid charge maximum power**` is at 'Inverter Max active power' level.
Causes overheating on the inverter!!!

Please limit the 'Batteries Grid charge maximum power' to 'Inverter Rated power' => this will immediately reduce inverter internal temperature to a secured temperature level.

### Describe your Huawei Solar Setup

Inverter Type: sun2000-6KTL-M1
Inverter Firmware version: latest
sDongle present: Yes
sDongle Type: sDongleA-05 (WiFi / Ethernet)
sDongle Connectivitiy: Ethernet
sDongle Firmware: latest
Power meter present: three phase
Optimizers Present: No
Battery: 1 x LUNA2000-SO 10kWh + 1 x LUNA2000-SO 5kWh
Battery Firmware version: latest
Huawei Solar integration version: latest

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

N/A

### Upload your relevant debug logs

```text
N/A
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [ ] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

heinemannj, 2024-11-01:
In Modbus Interface Definitions:

[image]

In [register_names.py](https://gitlab.com/Emilv2/huawei-solar/-/blob/master/src/huawei_solar/register_names.py):

[image]

In [registers.py](https://gitlab.com/Emilv2/huawei-solar/-/blob/master/src/huawei_solar/registers.py):

[image]

Temperature has been continuisly increased by from 60° to 63° within less than 2 minutes until I have reduced from 6600W to 6000W charge from grid power:

[image]

wlcrs, 2024-11-22:
Hi Joerg, 

I understand your worries. I take a good look into how everything currently works.

This integration follows the guidelines laid out by the modbus definitions pdf: the 'Batteries Grid charge maximum power' (register 47242) is governed by the register 47244.

Note that this register 47244 is also marked as 'writeable' in the definitions PDF, but that this integration does not have any code path that does that. I therefore assume that any unsafe value in that register 47244 is not caused by this integration.

I don't want to take any responsibility whatsoever on whether a configuration change of this inverter keeps it within safe operating conditions. It's up to the inverter to decide what charge value is within safe operating range.

Just for good measure:
# **I DO NOT TAKE ANY RESPONSIBILITY TO ANY DAMAGE THAT MAY BE CAUSED BY THE USAGE OF THIS INTEGRATION.**

# **USE AT YOUR OWN RISK.**

closed 2024-11-22 by wlcrs