wlcrs/huawei_solar #845: [Bug]: POWER_VOLT_AMPERE_REACTIVE was used from huawei_solar, this is a deprecated.

szczukot, 2024-11-07:
### Describe the issue

Logger: homeassistant.const
Source: helpers/deprecation.py:197
First occurred: November 1, 2024 at 6:17:49 PM (1 occurrences)
Last logged: November 1, 2024 at 6:17:49 PM

`POWER_VOLT_AMPERE_REACTIVE was used from huawei_solar, this is a deprecated constant which will be removed in HA Core 2025.9. Use UnitOfReactivePower.VOLT_AMPERE_REACTIVE instead, please report it to the author of the 'huawei_solar' custom integration`

### Describe your Huawei Solar Setup

Inverter Type: 
Inverter Firmware version: 
sDongle present: Yes / No
sDongle Type: sDongleA-05 (WiFi / Ethernet) / SDongleA-03 (4G)
sDongle Connectivitiy: WiFi / Ethernet / 4G
sDongle Firmware: 
Power meter present: three phase / single phase / no
Optimizers Present: Yes / No
Battery: LUNA2000-SO xxkWh / LUNA2000-S1 xxkWh (Released 2024) / LG RESU xxkWh / None
Battery Firmware version: 
Huawei Solar integration version: 

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
.
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

martinlindjarv, 2024-11-14:
It seems to have been resolved - but no release created afterwards.
(Look at pinned topic)

szczukot, 2024-11-14:
Yeah. I created also https://github.com/wlcrs/huawei_solar/issues/842 a few time ago.

closed 2024-11-22 by wlcrs