wlcrs/huawei_solar #847: [Bug]: all sensors Unavailable

Masadido, 2024-11-09:
### Describe the issue

Dar all,

I'm facing a really annoying issue with my SUN2000-8KTL-M1.
All sensors became "Unavailable". I didn't notice the problem directly but I'm guessing is might be related to software update.
I am using a connection to the inverter AP (192.168.200.1 port 6607) through a network bridge, which has been working perfectly so far. Ping 192.168.200.1 is working, so it doesn't seem to be a connection issue.
FusionSolar shows a normal operation of the SUN2000

[image]

### Describe your Huawei Solar Setup

Inverter Type: SUN2000-8KTL-M1
Inverter Firmware version: V100R001-02 (Software 100E001C00SPC161)
sDongle present: Yes 
sDongle Type: sDongleA-05 (WiFi / Ethernet) 
sDongle Connectivitiy: WiFi 
sDongle Firmware: V100R001C00SPC117
Power meter present: 
Optimizers Present: No
Battery: None
Battery Firmware version: 
Huawei Solar integration version: 1.4.1

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[config_entry-huawei_solar-2e4ab65d8f0bfd772810622cbe6015f1 (1).json](https://github.com/user-attachments/files/17687091/config_entry-huawei_solar-2e4ab65d8f0bfd772810622cbe6015f1.1.json)

[home-assistant_huawei_solar_2024-11-09T12-45-47.743Z.log](https://github.com/user-attachments/files/17687121/home-assistant_huawei_solar_2024-11-09T12-45-47.743Z.log)

[home-assistant_huawei_solar_2024-11-09T11-38-18.862Z.log](https://github.com/user-attachments/files/17687122/home-assistant_huawei_solar_2024-11-09T11-38-18.862Z.log)

### Upload your relevant debug logs

```text
See disgnostic files where I attached the logs
```

### Please confirm the following:

- [ ] I'm running the latest release of Home Assistant.
- [ ] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

Carletto86, 2024-11-10:
I also have the same problem with wired connection with dongle. It seems to have occurred in conjunction with Huawei's announcement of a change in the connection IP with the host (region04eu5.fusionsolar.huawei.com) dated 11/1/24. my dongle was already connected to the cloud via host and not via IP. The huawei app works fine

Pangel71, 2024-11-10:
I am having the same issue.
When I restart HA it works for some time (between 2 hours and 3 days), but then it stops again.
Inverter Type: SUN2000-8KTL-M1
sDongle Connectivitiy: WiFi

Filippo125, 2024-11-10:
I am having the same issue, connected to the inverter AP.
Inverter Type: SUN2000-6KTL-M1

kodacy, 2024-11-11:
I am having the same issue.
Inverter Type: SUN2000-4KTL-L1
sDongle Connectivitiy: WiFi
To recover connection I need to reconnect with fusion solar app directly to inverter AP and then reload the integration. until it fails again our reboot Home Assistant.

Mertensk, 2024-11-11:
Exactly the same here. 
I am looking for a way to downgrade my SUN20006KTL.
No succes yet. Not able to place a file on my phone for downgrading, nor a way to get in without a Huwaei smartlogger.
Maybe there is another way to get the data out of it?

cvanlith, 2024-11-12:
Is your WIFI/WLAN0 connection between inverter and ha-device still there? It was gone after I upgraded to core 2024.11.1.

Mertensk, 2024-11-12:
> Is your WIFI/WLAN0 connection between inverter and ha-device still there? It was gone after I upgraded to core 2024.11.1.

It was indeed missing. I reestablished the WLAN connection between the dongle in HA and the dongle on the SUN2000 and reloaded the integration.
Now everything is working perfect again! 
Thanks for the tip :)

theveninjeremy, 2024-11-13:
> Is your WIFI/WLAN0 connection between inverter and ha-device still there? It was gone after I upgraded to core 2024.11.1.

Hi 
Where can we check this? 
I faced the same issue. 
I removed the integration and now I cannot install it back. The connection failed. 
Thanks

Mertensk, 2024-11-13:
> > Is your WIFI/WLAN0 connection between inverter and ha-device still there? It was gone after I upgraded to core 2024.11.1.
> 
> Hi Where can we check this? I faced the same issue. I removed the integration and now I cannot install it back. The connection failed. Thanks

Hi,

I encountered the same issue after upgrading to core 2024.11.1. In my setup, I use a WLAN dongle on my NUC to directly connect to the WLAN dongle of my Huawei SUN2000 inverter. The dongle in Home Assistant is set to DHCP, and the connection works seamlessly, allowing all Modbus data from the inverter to be read.

You might want to check your WLAN0 connection between the inverter and your Home Assistant device. After the upgrade, I had to re-enter the SSID and password to re-establish the connection. Sometimes, network settings can be reset during an upgrade, and reconfiguring the Wi-Fi settings can resolve the connection problem.

You can check those settings in: settings, system, network, network interfaces

Hope this helps!

cvanlith, 2024-11-14:
> > Is your WIFI/WLAN0 connection between inverter and ha-device still there? It was gone after I upgraded to core 2024.11.1.
> 
> Hi Where can we check this? I faced the same issue. I removed the integration and now I cannot install it back. The connection failed. Thanks

For me the fastest way to check if the WIFI is still there, is to open Terminal and check the banner (or type ha banner).
It should state something like this:
`Welcome to the Home Assistant command line.`
`System information`
`  IPv4 addresses for wlan0: 192.168.200.2/24`
If there is no IP number for wlan0 your connection is gone.

jmichiel, 2024-11-20:
Had the same issue today after disconnecting my HA server from the network momentarily.
My dongle is connected via Ethernet cable normally, but I had WiFi setup initially.
I could see my device in the FusionSolar app, so it was working and had network, though pinging locally to the device IP (DHCP, but reserved IP) didn't work.
After removing the cable, the dongle reconnected over WiFi, and it became available again.
After reconnecting the cable, it reconnected over ethernet again, and it seems to keep working for now...

wlcrs, 2024-11-22:
This issue seems to be caused by a connection failure or an unstable connection to the inverter. This integration already has already implemented a lot of logic to automatically recover from connection failures or unstable connections. The three most important ones are:
- automatic reconnects if the connection to the inverter failed;
- each Modbus request is retried for up to 6 times, with increasing waits between each attempt to allow the connection to recover;
- after 3 failed Modbus requests the connection is reset to fix hanging connections.

Please understand that I cannot help you solve your connection failure or unstable connection. Consider [an alternative connection method](https://github.com/wlcrs/huawei_solar/wiki/Connecting-to-the-inverter).
If you are convinced that there is no connectivity issue, then I invite you to debug and fix the problem yourself and submit a pull request to the underlying [huawei-solar Python library](https://gitlab.com/Emilv2/huawei-solar). Your input would be greatly appreciated.

As there is nothing actionable for me to do in this issue, I'm closing it. You are welcome to open a topic in [Discussions: Connecting to the inverter](https://github.com/wlcrs/huawei_solar/discussions/categories/connecting-to-the-inverter) to get community support for your problem.

closed 2024-11-22 by wlcrs