wlcrs/huawei_solar #849: [Bug]: Efficiency always 100%

Isengo1989, 2024-11-12:
### Describe the issue

I want to calculate my input power per String with the A and V from the sensors, but efficiency is always 100%

### Describe your Huawei Solar Setup

Inverter Type: SUN2000-6KTL-M1
Inverter Firmware version: V100R001C00SPC159
sDongle present: Yes 
sDongle Type: sDongleA-05 (WiFi ) 
sDongle Connectivitiy: WiFi
sDongle Firmware: V200R022C10SPC108
Power meter present: three phase
Optimizers Present: Yes 
Battery: LUNA2000-SO 10kWh
Battery Firmware version: V100R002C00SPC119
Huawei Solar integration version: 1.4.1

[image]
[image]

### How do you connect to the inverter?

Please select your connection method

### Upload your Diagnostics File

[config_entry-huawei_solar-74c82f22992989ea0fcd1c66bfab8abc.json](https://github.com/user-attachments/files/17713825/config_entry-huawei_solar-74c82f22992989ea0fcd1c66bfab8abc.json)

### Upload your relevant debug logs

```text
2024-11-12 10:12:16.688 DEBUG (MainThread) [huawei_solar.bridge] Batch update of the following registers: storage_state_of_capacity, storage_running_status, storage_bus_voltage, storage_bus_current, storage_charge_discharge_power, storage_total_charge, storage_total_discharge, storage_current_day_charge_capacity, storage_current_day_discharge_capacity
2024-11-12 10:12:16.689 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760 with length 28 from slave 1
2024-11-12 10:12:17.486 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching TA2330385804_battery_data_update_coordinator data in 0.798 seconds (success: True)
2024-11-12 10:12:37.905 DEBUG (MainThread) [huawei_solar.bridge] Batch update of the following registers: state_1, state_2, state_3, alarm_1, alarm_2, alarm_3, pv_01_voltage, pv_01_current, pv_02_voltage, pv_02_current
2024-11-12 10:12:37.906 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000 with length 20 from slave 1
2024-11-12 10:12:38.660 DEBUG (MainThread) [huawei_solar.bridge] Batch update of the following registers: input_power, day_active_power_peak, active_power, efficiency, device_status, startup_time, shutdown_time, accumulated_yield_energy, total_dc_input_power, hourly_yield_energy, daily_yield_energy
2024-11-12 10:12:38.660 DEBUG (MainThread) [huawei_solar.huawei_solar] Sleeping for 0.048927 seconds before making next call.
2024-11-12 10:12:38.710 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064 with length 52 from slave 1
2024-11-12 10:12:39.488 DEBUG (MainThread) [huawei_solar.bridge] Batch update of the following registers: nb_online_optimizers
2024-11-12 10:12:39.488 DEBUG (MainThread) [huawei_solar.huawei_solar] Sleeping for 0.048343 seconds before making next call.
2024-11-12 10:12:39.539 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37201 with length 1 from slave 1
2024-11-12 10:12:40.216 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching TA2330385804_inverter_data_update_coordinator data in 2.311 seconds (success: True)
2024-11-12 10:12:40.225 DEBUG (MainThread) [huawei_solar.bridge] Batch update of the following registers: power_meter_active_power, active_grid_power_factor, active_grid_frequency, grid_exported_energy, grid_accumulated_energy, active_grid_A_power, active_grid_B_power, active_grid_C_power
2024-11-12 10:12:40.226 DEBUG (MainThread) [huawei_solar.huawei_solar] Sleeping for 0.040001 seconds before making next call.
2024-11-12 10:12:40.267 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37113 with length 25 from slave 1
2024-11-12 10:12:41.252 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching TA2330385804_power_meter_data_update_coordinator data in 2.462 seconds (success: True)
2024-11-12 10:12:46.688 DEBUG (MainThread) [huawei_solar.bridge] Batch update of the following registers: storage_state_of_capacity, storage_running_status, storage_bus_voltage, storage_bus_current, storage_charge_discharge_power, storage_total_charge, storage_total_discharge, storage_current_day_charge_capacity, storage_current_day_discharge_capacity
2024-11-12 10:12:46.688 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760 with length 28 from slave 1
2024-11-12 10:12:48.177 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching TA2330385804_battery_data_update_coordinator data in 1.489 seconds (success: True)
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2024-11-22:
The integration displays the values reported by the inverter. It does not interpret or compute it in any way.

My own inverter also always reports 100% efficiency 🤷‍♂️. I double checked the raw values received from the device when I saw this

closed 2024-11-22 by wlcrs

Isengo1989, 2024-12-20:
> The integration displays the values reported by the inverter. It does not interpret or compute it in any way.
> 
> My own inverter also always reports 100% efficiency 🤷‍♂️. I double checked the raw values received from the device when I saw this

Hmm sad. Would be cool if it could be calculated out of the inverters DC sum and overall yield.